wadpac/GGIR #585: Documentation on using cut-points from published literature

vincentvanhees, 2022-06-28:
Correct vignette description on cut-points for Esliger 2011, Phillips 2013, Fraysse 2020, Dibben2020:

Estimates are not realistic and I now think it should be:

threshold.lig = ((LightCutPointFromPaper_in_gmins/sampleRateInStudy)/60) * 1000
threshold.mod = ((ModerateCutPointFromPaper_in_gmins/sampleRateInStudy)/60) * 1000
threshold.vig = ((VigorousCutPointFromPaper_in_gmins/sampleRateInStudy)/60) * 1000
mvpathreshold = ((ModerateCutPointFromPaper_in_gmins/sampleRateInStudy)/60) * 1000

Unfortunately those studies were not very specific with their algorithm descriptions.

vincentvanhees, 2022-06-29:
and replace "InStudy" by "from paper" to clarify that it is the published study and not the data being processed

tomasvet, 2022-07-29:
Dear Vincent, Jairo, 
We are going to use GGIR using Dibben 2020 cut-points developed for heart failure patients. I think that the value of 40.6 mg they recommend for waist-worn accelerometers is realistic for this patient group (see the last paragraph of the Data reduction and analysis techniques section in their paper). After re-reading their paper several times, I understand that this value is an average rather than a sum metric as you suggested in the GGIR vignette (see equation 1 in their paper). The only pitfall is that they use the absolute value which, if I understand correctly, requires using ENMOa instead of ENMO. So in my opinion, I should use the setting do.enmoa= TRUE and acc.metric=”ENMOa”, but other than that I can use their cut-points directly without any scaling. Please let me know if I got something wrong. 
Thanks, 
Tomas

tomasvet, 2022-08-10:
Thanks for the answer and all your effort with GGIR.

closed 2022-08-10 by vincentvanhees