wadpac/GGIR #588: Movisens file info

vincentvanhees, 2022-07-01:
At the moment GGIR does not attempt to extract the ID or serial number from the movisens data. Probably because R package unisensR does not facilitate this at the moment. The XML file in the Movisens data indicates that ID and serial number are stored, so it would be good if we could try to extract this.

vincentvanhees, 2022-07-11:
This is not an ideal solution, but at least a way to enable ID extract for the time being. Next steps:

- [ ] Interact with unisensR package about getting this implemented on their side
- [ ] Test with various Movisens sensors.

jhmigueles, 2022-07-12:
I have now posted an issue in the [unisensR](https://github.com/Unisens/unisensR/issues/7) repository.

PR #609 (2022-07-26): Minor updates: movisense header recognition + activity log date format handling

Fixes #588 => for now by reading the xml file as a txt file, to be replaced by next release of uniseseR which will provide a more robust xml file read option.
Fixes #605 =>Improving activity log (GGIR part 2) date format handling and dealing with missing timestamps.

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [ ] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Updated or expanded the documentation.
- [... cut]
closed 2022-07-26 by vincentvanhees