wadpac/GGIR #606: g.getbout probably misclassifies bouted time as unbouted

jhmigueles, 2022-07-21:
When using `part5_agg2_60seconds = TRUE` and `boutdur.mvpa = c(1, 5, 10)`, I was expecting not to get any time into the **dur_day_MOD_unbt** and **dur_day_VIG_unbt** categories (the shortest bout lenght is equal to the epoch length, so any epoch above the moderate threshold should be classified as a bout of 1 minute). However, I get some minutes in the unbouted category, and I have noticed that all the **dur_day_MVPA_bts_1_5_min** have actually a duration of at least 2 min (and not 1 min as expected). I think it is related to the piece of code looking at the first and last minute of every bout meeting the bout.criter (g.getbout, bout.metric = 6).

vincentvanhees, 2022-07-23:
Sorry, I have been aware of this for a while but forgot to make the effort to document or address it.

If we would adjust the code to also allow for bout lengths that equal the epoch length then the concept of a bout may become unclear. Even if we document this behaviour it would still be hard to justify why we allow for it to happen.

The easiest solution I see is to define a bout as a time segment that lasts at least 2 epochs and meets the other criteria we specify for it. Further, I would then make it impossible for the user to specify a bout length that is equal or less than 1 epoch. For example, in the params_check function we could trigger an error when the user tries to specify this.

Just to clarify, this behaviour should only apply to the condition when bout length equals epoch length. If you specify 
boutdur.mvpa = c(5, 10) with an epoch length less than 5 minutes you should get  bouts that last for 5 minutes or longer. I tested that when I worked on the bout algorithm.

vincentvanhees, 2022-07-23:
... another option is that we automatically increase the bout length with 1 if the bout length equals the epoch length. This would then allow for using 1 minute bout length in part 2 when the epoch size is 5, but the code would automatically look for bouts of minimum length 2 minutes if the epoch length is 1 minute in part 5. We then only need to make sure this is reflect in the variable names of part 5. In that way the variable name will be consistent with the calculation.

commit ce75a4da (2022-12-07): fixes #606, and deprecates argument bout.metric and closedbout, integrating bug fixes by Jairo, and improving unit test. Further minor improvements to warning and stop calls related to myfun
PR #716 (2022-12-07): Bout detection improvements and simplification
@jhmigueles this PR integrates your revisions to the algorithm from PR #709, I copied your exact code without making modifications.

Further, I made the following complementary changes to other parts of GGIR:
- Arguments bout.metric and closedbout deprecated, as well as the code for all past bout metrics. I have removed them from both documentation (GGIR.Rd and g.getbout.Rd, and GGIR.Rmd) and from various functions.
- Unit test rewritten to focus on all possible scenarios where it goes wrong and no longer using random factors because that makes it [... cut]
closed 2022-12-07 by vincentvanhees