wadpac/GGIR #607: Error at beginning of run: 1 Error in if (dformat == 1 & is.mv == FALSE) { :    argument is of length zero

leightjt, 2022-07-25:
Hi,

I've just started using GGIR. I've started with an axivity ax3 .cwa file
I input:
GGIR(datadir="cwa_files",outputdir="results")

and I get the following error:
1 Error in if (dformat == 1 & is.mv == FALSE) { : 
  argument is of length zero

Any advice would be much appreciated.
Thanks,
Leigh

leightjt, 2022-07-25:
So I think I may have solved this issue. 
For any incompetent new users like me in the future, make sure .cwa is in lower case in the file name...

Thanks,
Leigh

closed 2022-07-25 by leightjt

vincentvanhees, 2022-07-26:
Interesting, how did it end up being in uppercase? OMGUI has always exported the files with lowercase extension I believe, but it would be good to know if there is route where files are exported in uppercase. If yes, then we could modify GGIR to also consider uppercase .CWA a valid extension for Axivity files.

vincentvanhees, 2022-07-28:
@leightjt please let me know your thoughts?

Ranudar, 2023-03-13:
Dear Vincent
I've just downloaded about 200 raw data files from the ActigraphCorph CentrePoint website, and all files have a ".CSV" file ending...

vincentvanhees, 2023-03-13:
For Axivity cwa/CWA format this was fixed in PR https://github.com/wadpac/GGIR/pull/744 

> I've just downloaded about 200 raw data files from the ActigraphCorph CentrePoint website, and all files have a ".CSV" file ending...

This issue does not related to ActiGraph. Please create a new issue and try to answer the questions I asked here: https://github.com/wadpac/GGIR/issues/737 If you do not know the answers then please contact the ActiGraph people first.

Ranudar, 2023-03-15:
Dear Vincent

I've given more information here:

https://groups.google.com/g/RpackageGGIR/c/AOOnKdA3FkM?pli=1

If you think I should open an issue on github, I'll of course do so.

Thanks a lot for your effort and help!
Christian