wadpac/GGIR #617: Update documentation

jhmigueles, 2022-08-16:
It would be good to update the documentation of the package:

- All arguments are defined in the same place in a consistent manner (help file of GGIR function). 
- Default options can be automatically identified in the documentation files, so that they do not have be revised if changed in the future.
- Include a vignette with the definition and default options of all the arguments in GGIR.

jhmigueles, 2022-08-21:
Hi @vincentvanhees, just finished updating the documentation of the GGIR arguments in GGIR.Rd plus doing a new vignette with all the parameters definition as we agreed. These two documents communicate to each other, so just updating the documentation in GGIR.Rd (in case it is needed in the future) will update the vignette. I had to include 2 new functions in GGIR for that (ShellDoc2Vignette and parametersvignette).

I think everything is fine but we may reduce the info in these documents by elliminating the arguments that are deprecated, and those which are not intended to be used (or very rarely used) by the users. Specifically:

Arguments we could remove:
- expand_tail_max_hours
- spherecrit
- minloadcrit
- chunksize --> this is reduced if needed within GGIR, right?
- dynrange --> the only scenario when we need this is with ad-hoc files, but we use rmc.dynamic_range for that
- rmc.file --> derived from datadir
- deprecated arguments

Additionally, to revise:
- We could make print.filename and printsummary hardcoded to TRUE and remove arguments,
- rmc.check4timegaps --> documentation says "the rest of GGIR is not designed for this", now this is considered within g.imputeTimegaps, right?

Once you tell me what to do with these arguments, I think we can merge this branch into master.

vincentvanhees, 2022-08-30:
Hi Jairo,

> Arguments we could remove:
> expand_tail_max_hours

Please keep. I added this argument earlier this year for a client. People kept asking me to allow for a way to include the last day of the recording in part 5 if the recording ends a few hours before midnight. I have added this argument to allow the user to expand the time series in part 1 in order to trigger the analysis of the last day in part 5, but ignores the meaningless last night from the part 4 and part 5 results. The main value of this is that you can still get the waking-hour estimates for the last day if you belief that the person fell asleep before or shortly after the end of the recording. I included a warning in the documentation that this procedure should be done with care. Therefore, I think it is important to keep documentation for this argument.

> spherecrit
> minloadcrit

Please keep in the technical documentation (man folder), but feel free to omit from the vignette.

> chunksize --> this is reduced if needed within GGIR, right?

Please keep. This is used by many studies who use GGIR on a machines with limited memory and/or who process large data files.

> dynrange --> the only scenario when we need this is with ad-hoc files, but we use rmc.dynamic_range for that

Please keep in technical documentation, but feel free to omit from the vignette.

> rmc.file --> derived from datadir

This argument is needed when testing the read.myacc.csv arguments, because then datadir is not specified. So, I think we need to keep it in the technical documentation, but feel free to omit it from the vignette.

> deprecated arguments

Agree. They can be omitted from the vignette.

> Additionally, to revise:
> We could make print.filename and printsummary hardcoded to TRUE and remove arguments,

With the submission of GGIRread to CRAN and the more thorough review it involved I learnt that they prefer that anything printed to the console can be easily turned off by the user. This means that all warning messages should be generated with `warning()` and all error messages with `error()`. Further, we need arguments like printsummary and print.filename to enable users to turn off the console message. At the moment there is no way for the user to turn off all the progress messages in the console 

So, I am wondering whether we could find a way to address all of this at the same time. For example:
- Do as you suggest, but then add Boolean argument `verbose` to allow user to turn off all progress messages, including printsummary and print.filename. We could then re-use this argument everywhere in GGIR for all normal non-warning, non-error messages.
- Make sure all warning/error messages are correctly given with `warning()` and `error()`.

> `rmc.check4timegaps --> documentation says "the rest of GGIR is not designed for this", now this is considered within g.imputeTimegaps, right?

There is still a [line](https://github.com/wadpac/GGIR/blob/c2eacd511e76d40920ada31b1e94d93c0d45f234/R/read.myacc.csv.R#L194) that relies on it, so I think we need to keep it for direct users of read.myacc.csv. However, you could remove it from the vignette as long as it remains part of the technical documentation.

jhmigueles, 2022-09-01:
- Regarding the arguments in the vignette and the GGIR help file, I will keep them all. The vignette is built to automatically copy all the arguments in the help file, so it would not be easy to remove arguments from only one place.  
- `verbose` argument: I think it's a great idea. Should I implement this in this branch or make a new issue/branch for this?

vincentvanhees, 2022-09-02:
> verbose argument: I think it's a great idea. Should I implement this in this branch or make a new issue/branch for this?

Feel free to do it in this branch, but I suspect that this could be more than a day of work and then it may be easier to treat it as a separate project. Initially, it would be good to focus on letting verbose control the console messages in the functions: GGIR, g.part1, g.part2, g.part3, g.part4, g.part5, g.calibrate, and g.getmeta. Once that works we may identify additional console messages that need to be controlled for.

Decision to be made is whether `verbose` should be a direct argument to the GGIR function, just like `mode, do.report, datadir, etc` or should it be part of a parameter object? => As verbose is such a generic parameter even relevant to someone not familiar with all the details of GGIR, I think having it as a direct argument is justified. We will then have to pass it to other functions.

closed 2022-09-20 by vincentvanhees