wadpac/GGIR #645: Double check non-wear time detection edges

jhmigueles, 2022-09-14:
Sometimes, mainly when long time gaps have been imputed, the nonwear time detection gets compromised, and the period of time prior the transition from nonwear to wear time seems to be misclassifed as wear. For example:
[image]
 
**Proposed solution**
After the nonwear detection, the edges of the identified nonwear periods could be double checked for additional windows of nonwear. In this double check, GGIR could look whether the sdcriter is met in 2/3 axes within window2 (long epoch size, default 15 min) instead of within window (default 60 min).

vincentvanhees, 2023-04-13:
As discussed, it would also be good to then use this moment to review the vignette paragraph about nonwear detection https://cran.r-project.org/web/packages/GGIR/vignettes/GGIR.html#53_Non-wear_detection
and where necessary expand/improve it.

jhmigueles, 2023-04-13:
Yes, this was my plan for this:

**1. Investigate if we can increase speed of nonwear detection**
Regarding this, I have investigated some alternatives to the for loops for the clipping and nonwear detection:
- _sapply_: it gets around 5% faster using sapply instead of for loops (but code is more difficult to read, maybe just for me)
- _vapply_: marginally better than sapply.
- _data.table package_: I am getting problems to aggregate the data using overlapping windows with the data.table package.

**2. Write unit tests for the new version of the algorithm**

**3. Extend documentation and vignettes**

Should I invest more time in improving speed or should we go ahead with the for loops for now? If so, I will write a unit test and then improve documentation/vignettes.

vincentvanhees, 2023-05-04:
> Should I invest more time in improving speed? If so, do you have any other suggestion? Or should we go ahead with the for loops for now? If so, I will write a unit test and then improve documentation/vignettes.

The algorithm has not become slower because of this enhancement, so let's skip this. I think we should consider this as part of a more general review of GGIR speed later this year.

closed 2023-05-10 by vincentvanhees