wadpac/GGIR #657: Error in `[<-`(`*tmp*`, i, 1, value = nightsummary.tmp$ID[1]) :    subscript out of bounds

GeoffreyWHarrison, 2022-09-26:
Hi,

GGIR shows an error when generating report part 4: 
Error in `[<-`(`*tmp*`, i, 1, value = nightsummary.tmp$ID[1]) : 
  subscript out of bounds

I've played around with what data I process but continue to get this same error at the same section. 

I saw a similar error in issue #300. 
Since you asked them for their session info here is mine:

R version 4.2.0 (2022-04-22 ucrt)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 19043)

Matrix products: default

locale:
[1] LC_COLLATE=English_Canada.utf8  LC_CTYPE=English_Canada.utf8    LC_MONETARY=English_Canada.utf8 LC_NUMERIC=C                    LC_TIME=C                      

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] GGIR_2.8-0    remotes_2.4.2

loaded via a namespace (and not attached):
[1] compiler_4.2.0    parallel_4.2.0    tools_4.2.0       curl_4.3.2        codetools_0.2-18  doParallel_1.0.17 iterators_1.0.14  foreach_1.5.2     data.table_1.14.2

I am using GENEActive .bin files. Recorded for variable numbers of days but interested (and restricted processing to) the first 14 days at 40Hz. No sleep diary and this was turned off. I changed almost nothing from the default argument values:

mode = c(1,2,3,4,5)
# mode=1
datadir = "D:/GENEActiv 2022 Data Trimmed"  
outputdir = "D:/2022 Final Data Output"
studyname ="Sleep Data Aug 2022"
f0 = 1  #start with first file
f1=0  #all files
# f1 = c()
GGIR(#-------------------------------
     # General parameters
     #-------------------------------
     mode=mode, 
     datadir=datadir, 
     outputdir=outputdir, 
     studyname=studyname, 
     f0=f0, 
     f1=f1,
     overwrite = FALSE, 
     do.imp=TRUE,
     idloc=1, 
     print.filename=FALSE,
     storefolderstructure = FALSE,
     #-------------------------------
     # Part 1 parameters:
     #-------------------------------
     windowsizes = c(5,900,3600),
     do.cal=TRUE, 
     do.enmo = TRUE,
     do.anglez=TRUE,
     chunksize=1,
     printsummary=TRUE,
     #-------------------------------
     # Part 2 parameters:
     #-------------------------------
     strategy = 1,
     # ndayswindow=14,
     hrs.del.start = 1,
     hrs.del.end = 1, 
     maxdur = 14,
     includedaycrit = 16, 
     L5M5window = c(0,24),
     M5L5res = 10,
     winhr = c(5,10),
     qlevels = c(c(1380/1440),c(1410/1440)),
     qwindow=c(0,24), 
     ilevels = c(seq(0,400,by=50),8000), 
     mvpathreshold =c(100,120),
     #-------------------------------
     # Part 3 parameters:
     #-------------------------------
     timethreshold= c(5,10),
     anglethreshold=5,
     ignorenonwear = TRUE,
     #-------------------------------
     # Part 4 parameters:
     #-------------------------------
     excludefirstlast = FALSE,
     includenightcrit = 16,
     def.noc.sleep = 1,
     # loglocation= "D:/sleeplog.csv",
     outliers.only = FALSE,
     criterror = 4,
     relyonguider = FALSE,
     sleeplogidnum = TRUE,
     colid=1, 
     coln1=2, 
     do.visual = TRUE,
     nnights = 14,
     #-------------------------------
     # Part 5 parameters:
     #-------------------------------
     # Key functions: Merging physical activity with sleep analyses
     threshold.lig = c(30,40,50),
     threshold.mod = c(100,120),
     threshold.vig = c(400,500),
     # excludefirstlast = FALSE,
     boutcriter = 0.8,
     boutcriter.in = 0.9,
     boutcriter.lig = 0.8,
     boutcriter.mvpa = 0.8,
     boutdur.in = c(10,20,30),
     boutdur.lig = c(1,5,10),
     boutdur.mvpa = c(1,5,10),
     timewindow = c("WW"),
     #-----------------------------------
     # Report generation
     #-------------------------------
     do.report=c(2,4,5))

## End(Not run)

vincentvanhees, 2022-09-29:
Can you send me v.vanhees@accelting.com the .RData file from the output subfolder meta/basic.

If I can replicate the error then maybe that helps me to see where the error comes from. Thanks

GeoffreyWHarrison, 2022-09-29:
> Can you send me ***@***.*** the .RData file from the output

Hi Vincent,

Thanks for the reply. I'm waiting to hear back from my superiors and ethics
about whether I can share the files. I have run the files successful on a
separate computer with Version 2.7-1. I get a similar error message when
running, since many of the files have participants not wearing the device
throughout the entire recording period. But in the older code version the
visual reports still get produced. Here are the error messages from the
older version that do not break the function from running:

Errors and warnings for Student02_Sess1_054607_2022-09-01
13-33-00.bin.RData$message
[1] "replacement has 0 rows, data has 311040"

$call
`$<-.data.frame`(`*tmp*`, "angle", value = numeric(0))

Errors and warnings for Student03_Sess1_054600_2022-09-01
14-13-46.bin.RData$message
[1] "replacement has 0 rows, data has 311040"

$call
`$<-.data.frame`(`*tmp*`, "angle", value = numeric(0))

vincentvanhees, 2022-10-24:
No closing this issue, feel free to re-open when you are able to provide more info on what caused the error or how to reproduce it.

closed 2022-10-24 by vincentvanhees