wadpac/GGIR #679: Make internal functions truly internal and remove them from namespace

vincentvanhees, 2022-10-11:
I have always made every new internal GGIR function user accessible based on the idea that this makes the code easier to navigate. However, this comes with a couple of downsides:

- It complicates restructuring internal functions, because we are bound to keeping the namespace consistent.
- The package reference manual has become very long, without a clear purpose as it is unlikely that users actually directly use the internal functions.
- All those internal functions need good up to date documentation, which is lacking for some of the internal functions. Maintaining this costs time.

**Proposal:**

Remove some of the internal functions from the namespace such as:
- [ ] g.dotorcomma
- [ ] g.downsample
- [ ] g.extractheadervars
- [ ] isfilelist
- [ ] g.weardec
- [ ] is_this_a_dst_night
- [ ] get_nw_clip_block_params
- [ ] get_starttime_weekday_meantemp_truncdata
- [ ] ismovisens
- [ ] g.readtemp_movisens
- [ ] iso8601chartime2POSIX
- [ ] POSIXtime2iso8601
- [ ] is.ISO8601
- [ ] g.part5.addfirstwake,
- [ ] g.part5.addsib
- [ ] g.part5.definedays
- [ ] g.part5.fixmissingnight
- [ ] g.part5.onsetwaketiming
- [ ] g.part5.wakesleepwindows
- [ ] g.part5.savetimeseries
- [ ] g.part5.handle_lux_extremes
- [ ] g.part5.lux_persegment
- [ ] g.sibreport
- [ ] g.part4_extractid
- ...

The assumption here is that none of these functions is directly used by any GGIR user.

**To do this:**
- Copy documentation from Rd file inside the function code, condense it, and make it a comment.
- Remove Rd file
- Remove function name from NAMESPACE file
- Make sure function is not called from within the documentation of any other function.
- Put clear identifier in function file name to help communicate to developers that it is an internal function
- Communicate in change log which functions have become internal.
 
I think it is easiest to do this in sub groups of functions, rather than trying to do it for all files in one go.

vincentvanhees, 2022-10-11:
I understand your concern @tarensanders However, I am not sure the solution you propose will work. When I do

`?ggplot2:::expand_range4`

I do not see the documentation for that internal ggplot2 function. It would seem that help functionality is only available if an Rd file exists, regardless of whether we use roxygen or not.

I suppose my aim is not so much to hide the function documentation or to make functions inaccessible but to make functions and documentation easier to navigate. In recent years GGIR has grown to have over 80 functions, while most users only need access to the `GGIR()` function (or `g.shell.GGIR`). 

Another solution is to review the naming of the 'internal' functions and make them more intuitive. If after that we still have difficulty navigating the functions then a follow up steps could be:
- Make a subset of the functions non-accessible to the user by removing them from the namespace. For example short functions with a simple purpose such as timestamp conversion.
- Split off some of the functionality to new R package(s) like I recently did with GGIRread.

vincentvanhees, 2022-10-11:
I have now made an attempt to map out how the function names could be updated. That said I think it probably wise to give this some reflection as we do not want to be changing function names again any time soon. Further, it may be good to do this  in phases, because it may be risky to change this many function names all at once.

I am using the part1, part2, part3, part4 and part4 such that functions strongly related to a certain part are alphabetically grouped to make them easier to find.

Current function name          |      New function name                    | Motivation
-------------------------------   | -------------------------------------- | -------------------------
check_myfun.R	                |	checkExtFunction.R                    | More consistent with other code related to external function embedding
g.abr.day.names.R	|	abbreviateDayNames.R      | Easier to read
g.detecmidnight.R	|	detectMidnights.R | Easier to read
g.dotorcomma.R	|	dotOrComma.R |  Easier to read
g.downsample.R	|	downSample.R |  Easier to read
g.extractheadervars.R	|	extractHeaderVars.R | Easier to read
g.getstarttime.R	|	getStartTime.R |  Easier to read
g.readtemp_movisens.R	|	readTempMovisens.R | Easier to read
read.myacc.csv.R	|	readCustomCSV.R  | Easier to read /pronounce
get_nw_clip_block_params.R	|	part1GetQCparams.R | Clarifies that this is used in part 1
g.applymetrics.R	|	part1ApplyMetrics.R                 | Easier to read
g.readaccfile.R | readFileChunk.R | Easier to read
g.analyse.avday.R	|	part2Averageday.R                      | Easier to read and clarifies link to part2
g.analyse.perday.R	|	part2Perday.R             | Easier to read and clarifies link to part2
g.analyse.perfile.R	|	part2Perfile.R                   | Easier to read and clarifies link to part2
g.conv.actlog.R	|	part2LoadActivityLog.R              | Easier to read
g.convert.part2.long.R	|	part2ConvertWide2Long.R | Easier to read and consistent with other functions in part 2
g.sib.det.R	|	part3SibExtract.R | Clarifies that this is used in part 3
g.sib.plot.R	|	part3SibPlot.R | Clarifies that this is used in part 3
g.sib.sum.R	|	part3SibSummary.R | Clarifies that this is used in part 3
HASIB.R	|	part3SibAlgosR | Easier to read
HASPT.R	|	part3GuiderAlgos.R | Better reflection of what function does
g.part4_extractid.R | part4ExtractID.R | Easier to read
g.create.sp.mat.R	|	part4CreateSibMatrix.R | Easier to read? Assumption is that user knowns abbreviation `sib`, which we use to refer to rest / candidate sleep periods and stands for sustained inactivity bout.
identify_level.R	|	part5ClassifyBehaviour | Easier to read
g.sibreport.R	|	part5SibReport.R | | Clarifies that this is used in part 3
g.part5.addfirstwake.R | part5Addfirstwake.R | Easier to read
g.part5.addsib.R | part5AddSib.R | Easier to read
g.part5.classifyNaps.R | part5ClassifyNaps.R | Easier to read
g.part5.definedays.R | part5DefineDays.R | Easier to read
g.part5.fixmissingnight.R | part5FixMissingNight.R | Easier to read
g.part5.handle_lux_extremes.R | part5LuxExtremes.R | Easier to read
g.part5.lux_persegment.R | part5LuxPerSegment.R | Easier to read
g.part5.onsetwaketiming.R | part5OnsetWakeTiming.R | Easier to read
g.part5.savetimeseries.R | part5SaveTimeseries.R | Easier to read
g.part5.wakesleepwindows.R | part5WakeSleepWindows.R | Easier to read
g.loadlog.R	|	loadSleepLog.R | Easier to read
g.plot.R	|	QCplot.R |  Clarifies that this is the plot in the qc folder
g.plot5.R	|	visualreport.R | Clarifies that this is the function linked to argument visualreport.

**Functions to stay the same:**

The following functions can stay the same because their name is either already clear enough or are more likely to have been used in research by which changing their name would cause consistency problems.

CalcSleepRegularityIndex.R, cosinorAnalyses.R, zzz.R, g.part1.R, g.part2.R, g.part3.R, g.part4.R, g.part5.R, GGIR.R, createConfigFile.R, datadir2fnames.R, applyExtFunction.R, check_params.R, extract_params.R, load_params.R, create_test_acc_csv.R, create_test_sleeplog_csv.R, g.analyse.R, g.calibrate.R, g.report.part2.R, g.report.part4.R, g.report.part5.R, g.shell.GGIR.R, g.fragmentation.R, g.getbout.R, g.getM5L5.R, g.getmeta.R, g.impute.R, g.imputeTimegaps.R, g.inspectfile.R, g.intensitygradient.R, g.IVIS.R, g.weardec.R, getfolderstructure.R, is.ISO8601.R, is_this_a_dst_night.R, isfilelist.R, ismovisens.R, iso8601chartime2POSIX.R, parametersVignette.R, POSIXtime2iso8601.R, ShellDoc2Vignette.R, tidyup_df.R, updateBlocksize.R, g.getmeta.R

**Function that needs to be split before it is given a new name:**

get_starttime_weekday_meantemp_truncdata.R

**The following functions have moved to GGIRread and will be deprecated soon:**

g.binread.R
g.cwaread.R
g.wavread.R

vincentvanhees, 2022-10-11:
@jhmigueles would be good to also get your thoughts on this (no rush, I see this more as a possible side-activity for the upcoming 6 months)

tarensanders, 2022-10-12:
You're correct, Vincent. You still need an Rd file for a function for the `?` to work. Marking it with the keyword internal just removes it from the documentation index rather than from the manual entirely. My mistake.

vincentvanhees, 2023-06-06:
New plan:
- [ ] Add keyword internal to many of the 'internal' GGIR functions, such that their documentation does not end up in the package documentation pdf. I am currently doing this.
- [ ] Gradually improve the naming of these internal functions as discussed above. It is probably best to do this gradually and not all at once, and to limit this to only the worst function names, because changing the names will complicate merging these changes back into branchs/forks of GGIR.

vincentvanhees, 2023-06-06:
Point 2 (keyword internal) is now in the master branch. Now closing this issue as I do not plan to do more work on it for the foreseeable future.

closed 2023-06-06 by vincentvanhees