wadpac/GGIR #689: Limit tail_expansion_max_hours to 5 hours max

vincentvanhees, 2022-10-21:
- [ ] It seems argument `tail_expansion_max_hours` makes it too easy for researchers to fake a long recording length and get biased results without even realising it. The documentation already warns about this, but documentation alone is not possibly enough. I think we should constrain `tail_expansion_max_hours` to a maximum of 5 hours. This means that if the assumption is made that the participant fell asleep on or before 7pm on the last recording day then synthetic data is appended if the recording ends after 7pm. Users will be prohibited from making this kind of assumption for any time-point earlier in the day.
- [ ] I believe I checked the functionality thoroughly, but it may be good to check again that the code excludes the last recording day if it ends at 6.55pm and not when it ends at 7.05pm.

jhmigueles, 2022-10-24:
I've got a quick question on this. So we want to only impute a maximum of 5 hours of data, which implies imputations from 7pm to midnight. What if `dayborder != 0`? Let's say `dayborder = 3`
- Do we want to keep the 5 hours as maximum? i.e., impute only if the recording ends after 10pm
- Or do we want to keep the 7pm as reference? i.e., in this case, we would allow to impute a maximum of 8 hours

jhmigueles, 2022-10-25:
Also, I'm thinking how we should consider this expanded time in the analysis. We cannot mark it as nonwear because it would not be used for the SPT detection. But if we leave it there, then we might have several hours analysed in part 5 with nonreal data that would be easily included in the datasets. So, I'm thinking we may mark it as -1 in `M$metalong$nonwearscore` and in `IMP$rout$r5`, so that it will be easy to use these data for the SPT definition and then skip it for the other analyses

vincentvanhees, 2022-10-27:
That is correct. The 8 is to make the expansion continue until 8am the next morning if dayborder = midnight to ensure the SPT window is detected. The length is probably not necessary, I guess 4 hours could also have worked but I did not see harm in setting it to 8. 

I think it is good to keep it this way and make it always 8 hours after dayboarder, regardless of the value of dayborder.

This in contrast to the 7pm threshold, which stays the same regardless of dayborder and tail_expansion_max_hours. So, dayborder and tail_expansions_max_hours will need to be used to assess whether 7pm threshold is met.

vincentvanhees, 2022-10-27:
Some users may want to constrain it to 8pm or to 9pm, because it is about the believe whether all study participants fall asleep at or before the time. In some studies 7pm may be save (pre-schoolers?), while in other studies 10 or 11pm may be chosen (older children). So, I now think it is best to keep this flexible.

How about changing it to a clock hour in the final night. For example, 19 to indicated 7pm and 22 to indicate 10pm? Maybe that makes the interpretation (and calculation) easier?

Whatever we do, I will have to communicate the change in functionality to the group I developed this for because they are using the current functionality.

PR #694 (2022-12-05): Issue689 tail expansion

Fixes #689 
Here a summary of this PR:
- `expand_tail_max_hours` now expects a time (in hours) at which the recording should end (or later) to expand the metashort and metalong data frames with synthetic data.
- The expanded time now is ignored in all the reports and visualizations (including visualisation_sleep.pdf and visualreports).
- Documentation of this argument is updated.

Things that might be improved:
-  I have been thinking about changing the name of this argument to make clear that it expects a time in hours, but I could not think of any good name. Any sugg [... cut]