wadpac/GGIR #691: Problems with calibration when the device is not worn during nights

jhmigueles, 2022-10-24:
Not sure if this is an issue or just something to be aware of and think if we should do anything. I have processed a bunch of files from accelerometers worn in the hip only during waking hours (so this might only apply to this scenary), so I did not impute the data (`do.imp = FALSE`).

**Issue**
Here a QC plot from one file (but most of the files presented this issue):
[image]
We can see some parts of the signal in which the accelerometer was not worn, but the acceleration is stuck at relatively high values (well over zero). If we zoom in:
[image]
Horizontal lines represent different options for light thresholds (i.e., 20, 30, 40).

Even seeing this, the calibration output looks good, so this can be potentially undetected by users:
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/jaihid/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/jaihid/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

cal.error.start | cal.error.end | n.10sec.windows | n.hours.considered | QCmessage
-- | -- | -- | -- | --
0.0352 | 0.00207 | 22877 | 120 | recalibration done,   no problems detected

</body>

</html>

**Implications**
- large amounts of likely sedentary behavior or sleep might be misclassified as light physical activity (maybe not a problem since most of it would be during sleep hours).
- If calibration is not performing well when the acceleration should be 0 (or close to it), this might be also affecting higher intensity levels of acceleration, right?

jhmigueles, 2022-10-24:
Small update, the QC plot looks better if using BFEN instead of ENMO, it seems that the band pass filter can fix this?

[image]

vincentvanhees, 2022-10-25:
There are probably multiple problems/solutions here:

**Problems:**
- Metric choice. As we know BFEN is less sensitive to calibration error => but changing metric does not offer an easy solution because absolute values are not comparable to ENMO
- Calibration procedure uses too much non-wear time, which skews the results.
- QC visualisation looks strange => not really a solution to fix only the visualisation
- With do.imp = FALSE we use the poorly calibrate values.

**Possible solutions:**
- Add option to impute by zero
- Modify g.calibrate to ignore near-duplicates.
- Add option to remove non-wear from day.

jhmigueles, 2022-10-28:
> Add option to impute by zero

Done, it works as expected, all the invalid segments are turned to zero (for all metrics except for EN which is set to 1).

> Modify g.calibrate to ignore near-duplicates.

I have tried rounding the dup argument to 4, 3, and 2 digits as we discussed. The effect in the calibration coefficients and the QC plots is trivial. This seems not to work. See for example:
[image]

> Add option to remove non-wear from day.

Followup question: do we want to do this only in the part 2 reports? Or also for part 4 and part 5 reports? If so, then we would need to consider the longest nonwear period of the day as SPT, right? This has the advantage that the users would also get Inactivity and Light PA estimates.

vincentvanhees, 2022-11-03:
> I have tried rounding the data to find duplicates to 4, 3, and 2 digits as we discussed. The effect in the calibration coefficients and the QC plots is trivial. This seems not to work. See for example:

Thanks for exploring this.
- Where can I find the code for this or is it not in a branch yet?
- We should use the rounded values for identifying duplicates and then go back to the original values for performing the auto-calibration. Did you do this?

> Followup question: do we want to do this only in the part 2 reports? Or also for part 4 and part 5 reports? If so, then we would need to consider the longest nonwear period of the day as SPT, right? This has the advantage that the users would also get Inactivity and Light PA estimates.

Sounds good to do this for part 4 and 5 too. I suppose this will be a big change to the code. Should we keep the current implementation as the default and do this as optional?

jhmigueles, 2022-11-03:
> Where can I find the code for this or is it not in a branch yet?

It is now in the [issue691_calibration](https://github.com/wadpac/GGIR/tree/issue691_calibration) branch.

> We should use the rounded values for identifying duplicates and then go back to the original values for performing the auto-calibration. Did you do this?

Yes, I only use the rounded numbers to detect duplicates and remove those values. But never modify the values themselves.

> Sounds good to do this for part 4 and 5 too. I suppose this will be a big change to the code. Should we keep the current implementation as the default and do this as optional?

I thought in a simpler way of doing this. The main purpose was to obtain the sedentary and light classifications in part 2, since when the device is not worn during sleep, we assume that all the wear time is awake and we do not need the SPT definition. So, I implemented the identify_levels function in part 2 (in `g.analyse.perday`) only when the device is not worn during sleep (there is a new argument for this). So, the standard processing in GGIR is not affected, this is an option only when the device is not worn during sleep.

I will make some more tests on this branch before doing a PR

vincentvanhees, 2023-12-06:
I am now closing this issue because it has been open for a long time and I am not sure it is still an issue.
The code now much better deals with idle sleep mode and deals with long episodes of constant sensor orientation.
Feel free to re-open if you think this is still an issue that needs attention.

closed 2023-12-06 by vincentvanhees