wadpac/GGIR #715: Expand main vignette

jhmigueles, 2022-12-07:
It would be good to expand the main vignette and revise the organization of the information that is presented there.

vincentvanhees, 2022-12-07:
If possible, try to keep the number of sections equal to avoid changing the section numbers, which would break vignette links people may have used. Although, if there is a clear need for additional sections then I guess giving up consistency is worth it.

vincentvanhees, 2022-12-08:
May I suggest that we first make an offline plan for this issue? FOr example, in shared Word document?

The risk is that we spend months on updating and expanding the vignette, but it may help to first have a map of what we aim for and priorities:
- To what extent does the section structure need to be re-order?
- Are there missing sections?
- Which sections are weak and need to be revised first?
- Are there parts that can be easily moved outside GGIR, e.g. to my website and then referenced from the vignette?
- Which of all these potential areas are most urgent? For example, which ones directly complement the GGIR training or which ones contain mistakes and are not clear to the user.

vincentvanhees, 2022-12-10:
Also, I think we need to do a careful typo check. For example, I just spotted that ActCR is written as "Ac tCR" in several places. It is as if the problem with the random spaces we had earlier is not resolved yet.

vincentvanhees, 2023-06-06:
Now closing this issue: We now plan to move substantial parts of the vignette to github.io documentation. Probably easier to create specific issues for that project then to have this issue still open.

closed 2023-06-06 by vincentvanhees