wadpac/GGIR #725: Interactive option to select directories if datadir/outputdir are not provided or do not exist

jhmigueles, 2023-01-19:
**Is your feature request related to a problem? Please describe.**
For new R users, the first challenge they find is usually the definition of the directories. It may be frustrating for them not being able to use GGIR just because of this. 

**Describe the solution you'd like**
I think providing the opportunity to interactively choose files/directories with utils::choose.files and utils::choose.dir would facilitate the introduction to GGIR. But I'm not sure if this should be extended for the definition of the log paths (sleeplog, activity log, data cleaning file...).

vincentvanhees, 2023-02-07:
Option 1:
As described above with pop-up window to select path.

Option 2:
Document better how to specify paths.

Option 3:
Document how to use
```
datadir = utils::choose.dir()
GGIR(data = datadir, ...)
```

vincentvanhees, 2025-09-22:
I am closing this issue. We expect users to make an effort to learn the basics of R, which includes specifying a file path. If user really want us to look after all their tiny concerns then I think it would be good if these users could come up with a proposal to maintain GGIR.

closed 2025-09-22 by vincentvanhees