wadpac/GGIR #727: Error in `[<-`(`*tmp*`, di, fi:(fi + 2), value = c(ID, fnames.ms3[i],  :    subscript out of bounds

nivlacdiep, 2023-02-06:
Hi Vincent,

Parts 1 - 4 all run smoothly, but in Part 5, I receive this error:
Error in `[<-`(`*tmp*`, di, fi:(fi + 2), value = c(ID, fnames.ms3[i],  : 
  subscript out of bounds

I suspect this is related to large amounts of non-wear time for our participants (perhaps similar to Issue #409 and [this Google thread](https://groups.google.com/g/RpackageGGIR/c/HwnIsaOtwJs)). This is an expected part of our study protocol, as we ask participants to wear an ActiGraph GT3X for 3-7 consecutive days before surgery, 7 consecutive days at 1 month after surgery, and another 7 consecutive days at 3 months after surgery; a total of ~20 days of wear time over a span of 4 months.

I'm using GGIR 2.8-2 and the GGIR command has all default arguments for now. 

Any help would be much appreciated. I'm happy to send over the metadata for a sample file.

Calvin

jhmigueles, 2023-02-18:
Hi Calvin, 

A sample file would be of help to replicate your issue and looking for a solution. Can you provide that? Part 1 metadata would be enough (files stored in the "meta/basic" folder). Thanks.

Best,
Jairo

nivlacdiep, 2023-02-18:
Hi Jairo,

Thanks for your reply. I've sent the file to your email. Let me know if you need any other files to replicate the issue!

Calvin

nivlacdiep, 2023-02-18:
Hi Jairo,

Here is the Part 1 metadata file from meta/basic. Let me know if you need
any other files to replicate the issue and take a look!

Calvin

jhmigueles, 2023-02-20:
Hi Calvin,

I'm sorry I can't see your file in any of my emails. Can you share a link to download the file?

Best,
Jairo

vincentvanhees, 2023-03-10:
@nivlacdiep could you try again with the latest GGIR version? We fixed a bug related to aligning part 3, 4 and 5 output, which may explain your issue.

nivlacdiep, 2023-03-11:
Hi Vincent and Jairo,

Unfortunately the same error comes up, even with the new version 2.8-6

Error in `[<-`(`*tmp*`, di, fi:(fi + 2), value = c(ID, fnames.ms3[i],  : 
  subscript out of bounds

What other files or metadata can I send to you for further investigation?

jhmigueles, 2023-03-15:
Hi, I tried to download the data some days ago but I had no access. Just tried again and still the same, can you either grant me access or send this to my email (jairo.hidalgo.migueles@gmail.com) so that I can have a look?

closed 2023-03-20 by vincentvanhees

vincentvanhees, 2023-03-21:
Hi @nivlacdiep this issue is closed because Jairo and I fixed it on Friday. To test it, please install master branch, re-start R and try again: `remotes::install_github("wadpac/GGIR")`. Let us know if the problem returns.

nivlacdiep, 2023-03-21:
Hi Vincent and Jairo,

I've tested the latest version and it does indeed work with my files. I appreciate all your efforts in addressing this issue.

Take care,

Calvin

PR #757 (2023-03-20): Issue727 bug part5 long files

Fixes #727 -> In long recordings, if there are many days excluded because of insufficient wear data, it might happen that the dsummary data frame generated in part 5 does not have enough rows to store the days (big discrepancy between total number of days recorded and days in which the spt window was calculated). In this branch, dsummary gets an extra row appended if needed to keep storing data.

This branch also includes @vincentvanhees work on the [issue727_speed_up_code](https://github.com/wadpac/GGIR/tree/issue727_speeding_up_code) branch.

Checklist before m [... cut]