wadpac/GGIR #737: g.inspectfile fails at detecting ".CWA" extensions

jhmigueles, 2023-02-23:
**Describe the bug**
There are some axivity files that are stored with the extension in uppercase (".CWA" instead of ".cwa"). This is a proble for g.inspectfile to detect that this file is a cwa.

vincentvanhees, 2023-02-23:
Do you know why there is a difference in capitals?
For example, did the researcher changed the extension or is it a change in the OMGUI software?
If OMGUI was changed, do we know why and whether the change in extension refers to a change in file format?
It is risky to assume that the .CWA can be processed as .cwa with more information on why it was changed.

jhmigueles, 2023-02-23:
I'm investigating that. For now, I have tried to fix this locally and it seems that the file gets processed and output looks normal just considering that .CWA and .cwa contain the same information.

Maylor8, 2023-02-23:
Hi both, I had the same issue just last week with a couple of files, they were skipped by GGIR so didn't make any meta files. I'm not sure I can replicate the circumstances which made them come out with capitals, but I will investigate to see if I can find the circumstance which lead to them being capitalised. 
On a side note for others in the meantime who may have this issue, I noticed that if you change this manually in windows you need to alter the extension via another random extension e.g. .CWA > .xyz > .cwa as changing straight from .CWA to .cwa in Windows 11 just reverted back to .CWA for me. 
.CWA is read by OMGUI just fine and the visual data summary seems from GGIR seems to match so changing .CWA to .cwa doesn't appear to have any underlying issues.
Thanks,
Ben

vincentvanhees, 2023-02-28:
@danielgjackson could you clarify?

danielgjackson, 2023-02-28:
The AX devices internally store the data on an original FAT filesystem with an "8.3" filename (without the later "long filename" extension, so it does not support mixed case): "CWA-DATA.CWA". If the file was directly copied off the device, the filename and extension would likely be all upper-case. 

If OmGui is used to download the file, it is saved to a file with a name based on a template (typically session ID and device ID) with a lower-case ".cwa" extension. 

The extension should be treated as case-insensitive, the contents are identical. I would only expect the non-mixed case "{.cwa,.CWA}" variations.

closed 2023-03-06 by vincentvanhees