wadpac/GGIR #745: make part5_agg2_60seconds work for fragmentation metrics only

Maylor8, 2023-03-06:
To generate fragmentation metrics, we need to include part5_agg2_60seconds=TRUE. However, this messes up all of the other metrics in the part5 summary outputs because all of it is calculated on 60s, not 5s data. 

Solutions:
1) Can this be made clear in the vignette for other users who may not realise this? 
OR
2) If possible, can this function work externally to the generation of the other traditional part5 outputs so we don't need to run it twice? one to get our daily summary variables based on 5s epochs, then again aggregated to generate fragmentation metrics?

A development somewhat related is also to have the function to generate csv outputs based on a 60s aggregate for timestamps in instances such as graphing purposes, when the actual summary metrics are made using 5s epochs.

vincentvanhees, 2023-03-10:
Thanks, great suggestion. I think both documentation and extra argument are needed.

Here, it may be helpful to think about a new input arguments to eventually replace `part5_agg2_60seconds` to make it all more flexible. For example one argument for each functionality, such as: `epochSize.part5.timeuse = 5` and `epochSize.part5.frag = 60`.

Maylor8, 2023-03-15:
Thanks Vincent, that would be an excellent solution if it was made that way.