wadpac/GGIR #753: Error produced in g.calibrate with csv file

wheelerb, 2023-03-13:
**Describe the bug**
Error in which(data[, 1] == 0 & data[, 2] == 0 & data[, 3] == 0) :
  'list' object cannot be coerced to type 'double'
Calls: GGIR -> g.part1 -> main_part1 -> g.calibrate -> which

By tracing through the GGIR source code, the source of the problem appears to be on line 169 of g.calibrate:
if (min(dim(S)) > 1) {
        data = rbind(S,data)
      }
During chunk 2, S is a character matrix, data is a dataframe with the timestamp column (column 1) having class c("POSIXlt", "POSIXt"). After the rbind, the timestamp column (column 1) becomes a list which gives the error shown above. 

**To Reproduce**
A csv file of about 2.5 GB in size (uncompressed) was the input data. The first two rows are:
         HEADER_TIMESTAMP     X     Y     Z
1 2000-01-02 12:30:00.000 0.704 0.460 0.191
2 2000-01-02 12:30:00.013 0.692 0.499 0.246

1. Sensor brand: '...'
2. Data format: ad-hoc csv file (mon = 0, dformat = 5)
3. Approximate recording duration is 7 days
4. Are you using a sleep diary to guide the sleep detection: YES / NO
5. Copy of R command used: 
GGIR(
  chunksize=1,
  mode=1:5,
  print.filename=TRUE,
  do.parallel=FALSE,
  overwrite=TRUE,
  #datadir="/data/UKBB_Matthews/NHANES_project/GGIR/Samples/Input_Dat",
  datadir="/data/wheelerwi/NHANES/results/2023/02-28/data2",
  outputdir="/data/wheelerwi/NHANES/results/2023/02-28/out",

  #=====================
  # read.myacc.csv arguments
  #=====================
  rmc.nrow=Inf, rmc.skip = c(), rmc.dec=".",
rmc.firstrow.acc = 2, rmc.firstrow.header=c(),
rmc.header.length = c(),
rmc.col.acc = 2:4, rmc.col.temp = c(),
rmc.col.time=1,
rmc.unit.acc = "g", rmc.unit.temp = "C",
rmc.unit.time = "POSIX",
rmc.format.time = "%Y-%m-%d %H:%M:%OS",
rmc.bitrate = c(), rmc.dynamic_range = c(),
rmc.unsignedbit = TRUE,
rmc.origin = "1970-01-01",
rmc.desiredtz = "Europe/London",
rmc.sf = 100,
rmc.headername.sf = c(),
rmc.headername.sn = c(),
rmc.headername.recordingid = c(),
rmc.header.structure = c(),
rmc.check4timegaps = FALSE,
rmc.col.wear = c(),
rmc.doresample=FALSE,
rmc.noise=13,
interpolationType=1,
PreviousLastValue = c(0, 0, 1),
PreviousLastTime = NULL,
epochsize = NULL,

  #=====================
  # Part 1
  #=====================
  do.report=c(2,4,5),
  imputeTimegaps=TRUE,
  do.anglez=TRUE,
  #=====================
  # Part 2
  #=====================
  strategy = 1,
  hrs.del.start = 0,          hrs.del.end = 0,
  maxdur = 9,                 includedaycrit = 19,
  qwindow=c(0,24),
  mvpathreshold =c(125),
  bout.metric = 4,
  excludefirstlast = FALSE,
  includenightcrit = 19,
  epochvalues2csv = TRUE,
  cosinor=TRUE,
  iglevels = TRUE, #this function calculates intensity gradient
  do.neishabouricounts=TRUE,
  winhr = c(5,10),
  M5L5res = 10,
  #=====================
  # Part 3 + 4
  #=====================
  def.noc.sleep = 1,
  nnights = 7,
  relyonguider=FALSE,
  outliers.only = TRUE,
  criterror = 4,
  do.visual = TRUE,
  excludefirst.part4=FALSE,
  HASIB.algo="vanHees2015",
  #=====================
  # Part 5
  #=====================
  f0=c(), f1=c(),
  excludefirstlast.part5=TRUE, 
  windowsizes=c(5,900,3600),acc.metric="ENMO", boutcriter.mvpa=0.9, 
  boutcriter.in=0.9, boutcriter.lig=0.9, storefolderstructure=FALSE, 
  threshold.lig = c(40,30), threshold.mod = c(100,125),
  threshold.vig = c(400), timewindow=c("MM","WW"), 
  boutdur.mvpa = c(1,5,10), boutdur.in = c(10,20,30), 
  boutdur.lig = c(1,5,10),
  desiredtz="", 
  dayborder=0, save_ms5rawlevels=TRUE,
  part5_agg2_60seconds = FALSE, save_ms5raw_format="csv",
  save_ms5raw_without_invalid=FALSE,
  data_cleaning_file=c(),
  
  #=====================
  # Visual report
  #=====================
  visualreport=TRUE
  )

7. Have you tried processing your data based on GGIR's default argument values? (YES) Does the issue you report still appear? YES 
8. Output to console:
 GGIR version: 2.8.2

   Do not forget to cite GGIR in your publications via a version number and
   Migueles et al. 2019 JMPB. doi: 10.1123/jmpb.2018-0063.
   See also: https://cran.r-project.org/package=GGIR/vignettes/GGIR.html#citing-ggir

   To make your research reproducible and interpretable always report:
     (1) Accelerometer brand and product name
     (2) How you configured the accelerometer
     (3) Study protocol and wear instructions given to the participants
     (4) GGIR version
     (5) How GGIR was used: Share the config.csv file or your R script.
     (6) How you post-processed / cleaned GGIR output
     (7) How reported outcomes relate to the specific variable names in GGIR
________________________________________________________________________________
 Part 1

Checking that user has read access permission for all files in data directory: Yes
Checking that user has write access permission for directory specified by argument outputdir: Yes
1
File name: 76217.csv.gz
P1 file 1

Investigate calibration of the sensors with function g.calibrate:

Loading chunk: 1 2Error in which(data[, 1] == 0 & data[, 2] == 0 & data[, 3] == 0) :
  'list' object cannot be coerced to type 'double'
Calls: GGIR -> g.part1 -> main_part1 -> g.calibrate -> which
In addition: Warning message:
In value[[jj]][ri] <- if (is.factor(xij)) as.vector(xij) else xij :
  number of items to replace is not a multiple of replacement length

**Desktop (please complete the following information):**
 - OS: Linux cn4297 3.10.0-862.14.4.el7.x86_64 #1 SMP Wed Sep 26 15:12:11 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
 - GGIR Version 2.8.2

**Additional context**
By tracing through the GGIR source code, the source of the problem appears to be on line 169 of g.calibrate:
if (min(dim(S)) > 1) {
        data = rbind(S,data)
      }
During chunk 2, S is a character matrix, data is a dataframe with the timestamp column (column 1) having class c("POSIXlt", "POSIXt"). After the rbind, the timestamp column (column 1) becomes a list which gives the error shown above. 

**Before submitting**
- [ ] Have you tried the steps to reproduce? YES 
- [ ] Do they include all relevant data and configuration? The data is probablt too large to include. 
- [ ] Does the issue you report still appear there?
- [ ] Have you tried this on the latest `master` branch from GitHub? YES

vincentvanhees, 2023-03-13:
I think this was fixed by the PR #743 that we recently merged. 
Can you try again with the GitHub master branch which equals GGIR version 2.8-6.
Thanks

wheelerb, 2023-03-13:
Thank you, the mater branch code did work.

closed 2023-03-13 by wheelerb