wadpac/GGIR #758: Ideas for future GGIR functionalities

vincentvanhees, 2023-03-21:
This is a list of functionalities that may be useful for some users, but require funding and/or time to develop. So, they are not currently on our timeline.

1. Split up recording, e.g. before and after an intervention.
2. Merge recording, e.g. Movisens, Empatica, Verisens and others tend to store data split per day.
3. Improved management of milestone data: reduce storage redundancy, efficient loading of data in memory, checksums to verify consistency, more platform independent, ....
4. Web-services surrounding GGIR
5. Tools to aid combining accelerometer data with other data such as GPS, GIS, physiological signals.

Suggestions or support with finding funding is welcome!

jhmigueles, 2023-04-12:
6. Add calculation of bouts for moderate intensity and vigorous intensity separately (only bouts of MVPA are provided at the moment, from [google group message](https://groups.google.com/g/RpackageGGIR/c/syTMFS-SVkg)).

vincentvanhees, 2024-01-14:
I am closing this issue as it is not of much practical value. If anyone wants this work to be done they should reach out and organise funding.

closed 2024-01-14 by vincentvanhees