wadpac/GGIR #762: Osteogenic index

tjrantal, 2023-03-29:
**Is your feature request related to a problem? Please describe.**
Skeletal loading ought to be estimated based on intensity and number of loading peaks based on the understanding of  bone mechanobiology. Energy expenditure-geared acitivty metrics do not optimally reflect the physiological underpinnings and hence an osteogenic index (and other impact peak count-based ideas) have been proposed for habitual skeletal loading estimation.

**Describe the solution you'd like**
I would like to include peak identification and intensity recording in GGIR and assign myself to implementing this feature.

**Describe alternatives you've considered**
I have implemented peak counting in our in house projects but would rather have the method available to the public.

**Additional context**
We've cited some of the pertinent literature in our related (somewhat recent, i.e. within the last five years) pubs, e.g.
doi: 10.1249/MSS.0000000000001770.
doi: 10.1016/j.bone.2020.115704
J Musculoskelet Neuronal Interact. 2022 Dec 1;22(4):514-523.

vincentvanhees, 2023-09-27:
If unsure how to start you may want to implement it first as an external function via: https://cran.r-project.org/package=GGIR/vignettes/ExternalFunction.pdf

Once that works integrating it in GGIR may be easier as you have been able to test whether it produces informative GGIR output.

vincentvanhees, 2024-01-30:
I am closing this issue now, feel free to re-open if you still plan to work on it.

closed 2024-01-30 by vincentvanhees