wadpac/GGIR #767: Indexing error

steffenhartmeyer, 2023-04-13:
**Describe the bug**
Calling the GGIR function (with default values) leads to indexing error:
Error in data_tmp[[1]][1:data_ln] : 
  only 0's may be mixed with negative subscripts

**To Reproduce**
Steps to reproduce the behavior.

1. Sensor brand: 'Axivity Ax3'
2. Data format: '.cwa'
3. Approximate recording duration '7' days
4. Are you using a sleep diary to guide the sleep detection: Not relevant for this bug
6. Copy of R command used: GGIR(datadir = datadir, outputdir = outputdir)
8. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem. Note that usually we are not only interested in see the error message in red, but all GGIR output to the console.

**Desktop (please complete the following information):**
 - OS: MacOS 13.1, MacbookPro M1 Pro
 - GGIR Version 2.9.0
 - R Version 4.2.2

**Additional context**
Add any other context about the problem here.

**Before submitting**
- [ ] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there?
- [ ] Have you tried this on the latest `master` branch from GitHub?
[image]

vincentvanhees, 2023-04-13:
I think the problem is the comma at the end of `outputdir,` Please change to:
```
GGIR(
  datadir = datadir,
  outputdir = outputdir
)
```

steffenhartmeyer, 2023-04-13:
Thanks for the fast reply! Unfortunately the problem persists without the comma.

steffenhartmeyer, 2023-04-13:
I think the call fails before creating the config file as there is no such file in the output folder.

vincentvanhees, 2023-04-13:
1. Are you providing the full path and not relative path as value for datadir?
2. The path specified by datadir is expected to have exclusively accelerometer files. Including non-accelerometer files or empty folder can confuse GGIR.
3. Based on your error message it seems the code has trouble recognising that this is not a Movisens data directory. Does your data directory have empty subfolders or xml files maybe?

steffenhartmeyer, 2023-04-13:
Ahh I did indeed specify relative file paths. It does work now with the full path. Thank you very much for your help!

closed 2023-04-13 by vincentvanhees