wadpac/GGIR #768: strategy = 3 takes angle data instead of acceleration

jhmigueles, 2023-04-13:
When `strategy = 3`, GGIR should look for the `ndayswindow` days with the highest acceleration values. Instead, it takes the angle data and look for the 6 hours with the highest average as starting point.

**Proposal**
- Use acceleration data, either `acc.metric` or first acceleration metric in `M$metashort`.
- Calculate full `ndayswindow`-length averages every `windowsizes[3]` windows (default = 1 hour [sliding window]).
- Use maximum value from the generated averages.

vincentvanhees, 2023-04-15:
> When strategy = 3, GGIR should look for the ndayswindow days with the highest acceleration values. Instead, it takes the angle data 

In the code I see that it takes the second column of metashort, which should always was the acceleration metric. It seems that as a result of either an R update or a GGIR update columns are no longer ordered as they always were. This would mean that any assumption in GGIR about column index would potentially cause problems. So, rather than focussing on this specific issue of ndayswindow,  it may be good to first investigate where the change in column order originates from and whether it also impacts on other parts of GGIR.

> and look for the 6 hours with the highest average as starting point.

I do not think this is what the code does. It runs an ndayswindow in steps of 6 hours through the data to find the most active ndayswindow.

vincentvanhees, 2023-04-15:
To follow up on what I wrote in my post above 20 minutes ago:

> So, rather than focussing on this specific issue of ndayswindow, it may be good to first investigate where the change in column order originates from and whether it also impacts on other parts of GGIR.

- I just went through the GGIR code and it seems the `ndayswindow` part in `g.impute()` is the only place in GGIR where we make a hard assumption about the column index of a variable in metashort. 
- The origin of the column order is in g.applymetrics where angle is simply calculated before ENMO. So, it sounds like that this bug originated when g.applymetrics was revised in recent years and the order of the calculation was altered.

I then agree on your solution for using `acc.metric`, but I am not sure on the second bit... Are you sure there is a mistake in the calculation of the rolling window? To me the probleem seems to be that it allows the non-zero ENMO values of non-wear to be used. So, I think it is better to temporarily turn them to zero for this part of the code. Note: Turning ENMO to zero for non-wearing everywhere in GGIR would be a major change and will undermine consistency of GGIR analyses for all who have used ENMO. So, I think it is best to treat that as a separate topic for discussion/investigation.

jhmigueles, 2023-04-17:
> I then agree on your solution for using `acc.metric`

Great, I have just implemented this in my local branch.

> Are you sure there is a mistake in the calculation of the rolling window?

You are right, the windows were correct. I have now turn nonwear epochs to 0 only here for this calculation (this does not affect the rest of the GGIR pipeline). But still, I think that the rolling window of 6 hours is probably too long? In the file I am using to check this, it results on this (start and end of the window could be improved):
[image]

I have two proposals for this:
1. Using the windowsizes[3] for the rolling window (default = 1 hour).
2. Focusing on the most active calendar days (starting and finishing at `dayborder`)

PR #774 (2023-04-28): Issue768 strategy3

Fixes #768 => Now strategy 3 and 5 uses `acc.metric` to look for the most active `ndayswindow`. Strategy 3 would look for X 24-h blocks, and strategy 5 would look for X complete calendar days.

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [x] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Updated or expanded the documentation.
- [x] Updated release notes in `inst/NEWS.Rd` with a user-readable summary. Please, include references to relevant issues or PR discussi [... cut]
closed 2023-04-28 by vincentvanhees