wadpac/GGIR #770: sep argument for reading and storing files

jhmigueles, 2023-04-19:
GGIR users with different default settings to separate lists in their machines would benefit from a general `sep` argument to be used for reading/storing csv files.

jhmigueles, 2023-04-30:
Update on this: For now, I have looked for parts of the code in which a csv file is read or written by GGIR and implemented the argument `sep`. This includes reading accelerometer raw data csv files, read.myacc.csv, and all the logs that can be read by GGIR (sleeplog, activity log, data cleaning file, etc).

I'm afraid that the separator might not be consistent in some countries in Europe, for example, the separator for the logs might be by default ";" in Spain, but the separator used when downloading a csv raw data file from other brand-specific software might be hardcoded to ",". I know that actilife uses the machine default to export csv files, but I have no idea about others (e.g., axivity csv, genea csv, adhoc csv files exported by other brands). Do you think that including this argument can be problematic in these cases?

vincentvanhees, 2023-05-01:
Can we keep it simple and only do this for writing csv files? For example, with a new argument `sep_csv_write`.

For reading accelerometer- or diary csv files I think we should not change anything, because this has been working fine for years. Changing it will introduce risk for many errors and user-questions.

jhmigueles, 2023-05-02:
Sounds good for reading accelerometer files, I think it is better not to change that. For reading diaries, GGIR actually has this option for the sleeplog (`sleeplogsep`), but not for the other diaries. Should we be consistent and offer the option for all the diaries? Maybe `sep_csv_read_logs`?

vincentvanhees, 2023-05-04:
Good idea, so conclusion:

For now we only do:
- sep_logs
- sep_reports

Maybe in the future:
- sep_acc => for csv accelerometer data

jhmigueles, 2023-05-22:
Update on progress in this branch.

**sep_logs** implemented in: 
- sleep log
- activity log
- csv with calibration coefficients
- TimeSegments2ZeroFile
- Data cleaning file

**sep_reports** implemented in: 
- reports stored in results folder (including data_quality_report and all part 2, 4 and 5 reports)
- Timeseries stored in csv in part 2 and part 5
- Legend file for timeseries in part 5
- SIB report

**To think about**
g.plot5 loads the previously-generated reports, which used to be stored with sep = ",".  If now the user can change this to any other sep, g.plot5 will not be able to load such reports.
`sep_logs` is not appropriate to handle this as the reports are not logs, and `sep_reports` is intended to be
used when writing new reports, not when reading reports... So, what should we use? Or do we need another sep for this?

**Maybe in the future:**
- sep_acc => for csv accelerometer data
- sep_config?

vincentvanhees, 2023-05-25:
Thanks, yes, sep_config would be good too.

PR #791 (2023-06-18): Issue770 sep arguments for reading csv files and storing reports

Fixes #770 --> Includes sep_reports, sep_logs, and sep_config to read csv files (logs and config file) and to store reports.

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [ ] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Updated or expanded the documentation.
- [x] Updated release notes in `inst/NEWS.Rd` with a user-readable summary. Please, include references to relevant issues or PR discussions.
- [ ] Added your  [... cut]
closed 2023-06-18 by vincentvanhees