wadpac/GGIR #839: Implement DFA analysis

vincentvanhees, 2023-07-11:
Implement DFA analysis based on code provided by Ian

vincentvanhees, 2024-01-31:
Hi @iandanilevicz and @victormesquita40

This is a heads-up that I am now implementing the DFA, SSP and ABI functions you wrote last year:
- Even though you are listed in the code as author, I will also add you as co-author to the git commit such that it is clear from the git history that you did the work. You will probably receive a notification about this.
- I am preparing all this in a git branch which I will turn into a pull request later today. I will not merge the pull request without Ian's approval (we will meet on Friday).
- I have made minor changes to the exception handling, which I will explain in the pull request. The algorithm itself is the same as how you provided it to me.

victormesquita40, 2024-01-31:
Hi people,

I made some promises to Ian about the code that I would review, but unfortunately, due to the lack of time, I couldn't.

Ian has some projects with Paulo Canas, who spoke highly of Ian to me, so he has my authorization to update the DFA package and its functions.

Please keep me updated on the progress of the versions.

Regards,

Victor

vincentvanhees, 2024-08-22:
Close this issue, because this is covered by https://github.com/wadpac/GGIR/issues/955 and will be merged paper has been written.

commit 17bea1f2 (2024-02-02): add DFA functionality, fixes #839
closed 2024-08-22 by vincentvanhees