wadpac/GGIR #881: One epoch bias in onset and wakeup part 5

vincentvanhees, 2023-08-11:
@jhmigueles your minor fix to the timing of the segments in part 5 (see #838) seems to have affected onset and wakeup timestamps in part 5. The sleep onset and waking time in part 5 are now one epoch later than in part 4. The epochs in part 4 are correct, so part 5 needs to be fixed 

- Either the addition of epoch change in part 5 needs to be reverted to how it was and fixed in a different way or the identification of the onset and wake-up time needs to be subtracted by 1 epoch.
- The unit-tests missed this, so they also need to be improved.

jhmigueles, 2023-08-21:
@vincentvanhees, I run the master branch on one of my test files (using defaults) and the timestamps in part4_nightsummary and part5_daysummary are identical. Can you share the file in which you are getting the different timestamps?

vincentvanhees, 2023-08-29:
Sorry, my mistake. I forgot that timestamps were rounded to nearest minute in part 5 when aggregating to minute.

closed 2023-08-29 by vincentvanhees