wadpac/GGIR #904: Interpretation of LUX variables

netiony, 2023-09-15:
Hello,

I have been working with GGIR to analyze our data, and we are close to publishing. I wanted to ask a bit about clarification of the LUX variables. We are getting M10_mean_peakLUX/M10_max_peakLUX variables as significant, but we aren't sure how to interpret that biologically. Based on the manual, they are the mean or maximum of the peak ambient light during the most active 10 hour period. Do you have any insights into the interpretation of these variables? Does this have to do with what time they get up in the morning, or how often they are outside? 
Thanks!

vincentvanhees, 2023-09-15:
> Based on the manual

Please clarify what part of the documentation you are referring to. A url with paragraph number or page number would help.

netiony, 2023-09-15:
Hello!

To clarify I am referring to the manual here: https://cran.r-project.org/web/packages/GGIR/vignettes/GGIR.html#5101_MXLX

In section 5.10.1 it mentions the LUX variables from GENEactiv devices: "Within GGIR part 5 MXLX is calculated per window, and if used in combination with the GENEActiv accelerometer brand LUX estimates per LX and MX are included."

Hope this clarifies my question!

vincentvanhees, 2023-09-27:
> Based on the manual, they are the mean or maximum of the peak ambient light during the most active 10 hour period. Do you have any insights into the interpretation of these variables? Does this have to do with what time they get up in the morning, or how often they are outside?

Below I have typed out description of all decisions involved, which hopefully also answers your question. Action point for me is now to add this to the vignette:

GGIR calculates the peak lux per long epoch at a default resolution of 15 minutes, which can be modified with argument `windowsizes`. We look at peak light to get a more reliable estimate of light exposure per time point compared with taking the average. GGIR part 2 visualises the LUX values in the qc plot, but does not summarise the data. In part 3 and 4 LUX is not used for sleep classification. In part 5 we calculate the mean and maximum of the peak LUX per epoch across all waking hours of the day. Here, the mean (peak per epoch) LUX would then indicate average light exposure per time segment, while max peak would indicate the maximum light exposure per day. Further, we calculate the max and mean peak LUX per most active consecutive X hour of the day. This intended to offer an alternative to LUX exposure during waking hours which relies on correct sleep classification. LUX exposure during M10 may be seen as an alternative if you are unsure whether you can trust the sleep classification in your dataset.

I hope this answers your question. If you have any questions or suggestions for improving the above text then let me know.

commit 74ad5eb4 (2023-10-05): Adding paragraph on LUX sensor data to vignette as there was no documentation, fixes #904
closed 2023-10-05 by vincentvanhees