wadpac/GGIR #943: Date-oriented strategy to define study protocol

jhmigueles, 2023-10-12:
Defining certain study protocols in GGIR might be challenging with the current strategies available. Introducing a date-oriented strategy could enhance flexibility to accommodate various scenarios.

Here, users would be able to provide a CSV file containing ID, first and last date for the study (only start date would also be supported). GGIR would then utilize this information to mask the data. Additionally, this strategy should be able to interact with `hrs.del.start` and `hrs.del.end` to skip specific portions of the first and last days.

I can think of 2 options for this:
1. Make the argument `strategy` able to use a character string as input indicating the path to the CSV file (similar to how `qwindow` works).
2. Introduce `strategy = 6` and a new argument, for example, `data_trimming_file` or `data_masking_file`

Option 1 prevents adding new arguments, while option 2  proposes arguments that are probably easier to understand and document.

vincentvanhees, 2023-10-13:
Maybe call it `study_dates_file` as this is specific about dates, the term `data` would be less specific.

I think there is a third and fourth option:
3. New argument `study_dates_file` and if it is specified then ignore strategy.
4. New argument `study_dates_file` and if it is specified then use it on top of what `strategy does`. For example a recording that lasts from 12/5 to 25/5 with strategy 3 would look for the most active nwindowsays between those dates. However, for strategy 2 and 4 I think this is less meaningful, so maybe only relevant to make this work for strategy 1, 3 and 5.

Option 1: Using both numeric and character values does add some extra complexity
Option 1 and 2: prevent us from combining strategy 3 with the new functionality.
Option 3: Would be the quick solution to those who exactly know when the wear period was. 
Option 4: Would be the most elaborate approach where we facilitate a variety of combinations.

What do you think?

On a separate note - Argument `strategy` 's name was poorly chosen by me as it not clear what it does. Ideally it should be renamed to something more intuitive. How about: `data_masking_strategy`. We could then keep `strategy` for consistency but update all example and documentation to focus on the new variable name.

jhmigueles, 2023-10-16:
I'd go for option 4 because it could make the task easier in more use cases. Going for a quicker fix now would increase the chances of having to revisit strategies in the future. Do you agree?

I agree on `data_masking_strategy` as the new name for `strategy`, sounds good. Is it ok if internally we keep using `strategy` and I just overwrite `strategy` with `data_masking_strategy` in `check_params`? I would also include some checks to make sure the use of `strategy` and `data_masking_strategy` by the users makes sense.

vincentvanhees, 2023-10-16:
> I'd go for option 4 because it could make the task easier in more use cases. Going for a quicker fix now would increase the chances of having to revisit strategies in the future. Do you agree?

I agree

> Is it ok if internally we keep using strategy and I just overwrite strategy with data_masking_strategy in check_params?

I think it is best to facilitate both for a while as a lot of people will be familiar with `strategy`. Maybe best to first allow both without giving a warning, and in six months time to start showing a warning when the user uses `strategy`. Internally you can indeed switch to the new name.

PR #948 (2023-11-03): Issue943 strategy dates

Closes #943 => added new function that allows to provide a csv file indicating the first and last date for the study protocol. This works on top of the strategy, and therefore can be combined with any of the current strategies in GGIR.

**Additionally:**

- Argument `strategy` renamed as `data_masking_strategy` (both options work for now).
- New function `check_log()` to avoid duplication of code when checking ID and date formats in activity log and study dates file. This could also be applied to sleeplog, but I didn't do it because we don't have a sleeplog_dateformat  [... cut]
closed 2023-11-03 by vincentvanhees