wadpac/GGIR #945: Change default long epoch size from 900 to 60?

vincentvanhees, 2023-10-15:
With the upcoming 3.0-0 release we will change the non-wear detection algorithm (argument nonwear_approach from default 2013 to 2023). It may be good to use this opportunity to also change the default second value of argument `windowsizes`, and go from c(5, 900, 3600) to c(5, 60, 3600). This then gives a higher resolution of non-wear detection, as well as higher resolution lux and temperature data.

Probably good to discuss first before I implement this...

vincentvanhees, 2023-10-16:
@jhmigueles do you agree? I could do this and then prepare 3.0-0 submission to CRAN in upcoming days.

jhmigueles, 2023-10-16:
I was doing some tests on this. Using 60 instead of 900 seconds doubles the processing time for part 1 in my case (I have checked with files of different size and it takes around double the time in all my checks). Other than that everything looks good, but I also noticed that the data aggregation in g.plot to export the plots_to_check_data_quality.pdf would use 60 seconds and the pdf size is larger as a result, but this is maybe less relevant.

vincentvanhees, 2023-10-16:
Update: I will keep the default to avoid risk of causing trouble this short before the CRAN submission. For most users 900 should be fine.
 
However, I have now prepared some updates to speed up detect_nonwear_clipping.R, which should ease `windowsizes = c(5, 60, 3600)`. To achieve this I changed the code such that it uses a 5 Hertz version of the original raw signal for non-wear detection. I do this by only taking the first value of every 0.2 seconds in the data. I do this because 5 Hertz should be enough to detect whether the sensor is worn and it makes the code a lot faster when using shorter long epoch lengths.

closed 2023-10-16 by vincentvanhees