wadpac/GGIR #952: Overlapping of 1 second when combining chunks of data in g.getmeta

jhmigueles, 2023-10-24:
While working on the issue #571, I have noticed that there is an overlap of 1 second in every block that is read using gt3x files (now testing in files from other brands). 

`g.getmeta` selects the data to be analyzed in each chunk and stores the remaining data to be appended to the next chunk in the object `S`. The last second in `S` is repeated as the first second in the next chunk, and, when combining the objects, there is a repetition of this second in the time series. 

Below a plot to visualize this showing the part of the timeseries in which the chunks have been appended:
[image]

jhmigueles, 2023-10-24:
**Update:** It seems this only occurs with gt3x files

vincentvanhees, 2023-10-25:
gt3x data is read by selecting it in chunks (batch_begin, batch_end) which length is defined in seconds. So, I maybe something goes wrong with incrementing the start and end of each chunk. For example, code does 1-11, 11-21, not realising that 11 is included in both?

PR #958 (2023-11-05): Issue571 log idle sleep mode actigraph

Fixes #952 => indexing of the chunks to read actigraph gt3x files revised so that there is no overlap of one second across chunks.
Closes #571 => now function `g.imputeTimeGaps` logs the number of gaps found and the total length of the gaps in minutes. This will facilitate not only logging when ActiGraph files had the idle sleep mode activated but also logging time gaps in any brand or file format processed. The logging of this information is done as the recent implementation with axivity files so that the `data_quality_report.csv` shows the number of im [... cut]
closed 2023-11-05 by vincentvanhees