wadpac/GGIR #961: g.part1( )

xiaoxiongjiangtang, 2023-11-02:
**Describe the bug**
When I input the codes as follows :
g.part1( datadir="C:/Users/13385/Desktop/GGIR_demo/62161", metadatadir="C:/Users/13385/Desktop/GGIR_demo/Results",f1=1)

The GGIR shows error :
Error in getbrand(filename, datafile) : 
Error processing GT3XPLUS-AccelerationCalibrated-1x8x0.NEO1G62402505.2000-01-08-12-30-00-000-P0000.sensor.csv: unrecognised file format.

The 62161 file is a folder containing raw data downloaded from NHANES.

How can I solve this error? Thank you very much.

**Desktop (please complete the following information):**
 - OS: Windows
 - GGIR Version: 3.0-0

xiaoxiongjiangtang, 2023-11-02:
Here is my RStudio version:
RStudio 2023.06.0+421 "Mountain Hydrangea" Release (583b465ecc45e60ee9de085148cd2f9741cc5214, 2023-06-05) for windows
Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) RStudio/2023.06.0+421 Chrome/110.0.5481.208 Electron/23.3.0 Safari/537.36

vincentvanhees, 2023-11-05:
Do you have the `.gt3x` files? GGIR is designed to facilitate ActiGraph files in `.gt3x` format and older conversions of .csv to `.gt3x`. If you are working with a newer data formats then GGIR may not be able to handle those yet.

Have you tried using the `GGIR()` function as explained in `https://cran.r-project.org/web/packages/GGIR/vignettes/GGIR.html#23_GGIR_shell_function`? Using `g.part1` should be possible but I would recommend working with `GGIR` as this already takes care of setting up the right folder structure.

vincentvanhees, 2023-11-23:
I am closing this issue now, feel free to re-open, if you have any update.

closed 2023-11-23 by vincentvanhees