wadpac/GGIR #962: Shell script fails with error message: "'from' must be a finite number and "$call seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) * ws3)), by = ws3)"

charleslucenap, 2023-11-02:
Hello all,

 I've got a database with data from 1014 participants who wore the GENEA Accelerometer. 

In GGIR, after part 1, the R program presents the following: "$call seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) * ws3)), by = ws3)

In addition: Errors and warnings for 011057_01-04-2014 16-16-03.bin$message [1] "'from' must be a finite number

After that, in part 4, the R program generates just 860 files instead of 1014 (original). Probably, I think that these differences are about this "error". 

In part 5, I received the following message as well: "Error in 1:max(summarysleep_tmp$night) : "

I am using the GENEActiv, as well as .bin file. Curiously, these data were performed with other cut-offs in the past, and everything works well. However, I'm updating the cut-off and identifying these errors.

I've been looking at other posts that mentioned idloc and colid, however, I tried to modify that but doesn't work. Are there potential solutions? If yes, please, let me know.

Many thanks for this help.

Charles.

vincentvanhees, 2023-11-05:
Issue come with templates, which you ignored. Please stick to the template.

closed 2023-11-05 by vincentvanhees

charleslucenap, 2023-11-06:
Sorry! I created other topics following these instructions. Please, have a look if this is correct.