wadpac/GGIR #963: Create internal data with part 1 output to run tests

jhmigueles, 2023-11-03:
**Adapted from @vincentvanhees [comment](https://github.com/wadpac/GGIR/pull/948#discussion_r1380057441)**

Running part 1 in multiple unit tests slows down package testing. 

Proposal 1: Include the part 1 output in internal package data and then re-use that for every unit test that is not directly concerned with testing part 1 but depends on part 1 output. Storing the full part 1 output takes up too much space. In https://github.com/wadpac/GGIR/pull/956 I solved it by only storing metalong, object. What we could do is that we use that data to reconstruct M$metalong as I do in that branch and next we could simulate a matching metashort.

Proposal 2: Another approach may be to make sure that part 1 is run once when running the tests and that its data remains available for other unit tests. I do not know how to do this but it may be worth investigating.

vincentvanhees, 2023-11-08:
@jhmigueles 

Another approach may be to combine unit tests more efficiently. We currently have 50 separate unit tests which is a lot.
For example, we could pool them in a way that they are structured with multipel unit test inside a single test file as:

```
GGIR(mode = 1,...)

test_that("functionality A that depends on part 1 output", {
 GGIR(mode = 2:5,...)
}
test_that("functionality B that depends on part 1 output", {
}
test_that("functionality C that depends on part 1 output", {
}
```

this may save a lot of computing time as multiple tests can then utilise the initial part 1 output and help to condense the number of unit test files into clear groups of similar functionality.

For example, we could have the fulling unit test files:
- All of GGIR (test_chainof5parts.R)
- All function in part 1.
- All function in part 2
- All function in part 3
- All function in part 4
- All function in part 5

Well, maybe this is too ambitious, but we could try to consider that while also looking for ways to internally store part 1 data.

jhmigueles, 2023-11-28:
@vincentvanhees I investigated one option to create part 1 milestone data to be used in the tests. The [`create_part1_meta`](https://github.com/wadpac/GGIR/blob/issue963_simplifyTests/R/create_part1_meta.R) function uses the data objects that were already stored in the data folder in the package and repeats the data there to meet the desired duration of the recording (defined with `Ndays`). The function comes with the possibility to add or not to add light column to speed up the generation if not needed.

As this function takes data that were already stored in the package, it doesn't affect the size of the package. I tested this in [test_lightPart5](https://github.com/wadpac/GGIR/blob/issue963_simplifyTests/tests/testthat/test_lightPart5.R), and locally in my laptop the test takes ~35 seconds in the master branch version, and ~15 seconds using this new approach.

vincentvanhees, 2023-11-30:
I think that is a good idea as it does not require major changes to the existing tests.

vincentvanhees, 2024-04-04:
As discussed today, we will explore:
1. Would be good to investigate whether create_test_acc_csv can be speeded up
2. Explore ways to store test files in a separate package or online and download from there, e.g. zenodo? (similar to how read.gt3x obtains its test file). If this is possible then add:
  - Multiple Actigraph csv files as we currently generate in the tests
  - Real files from various sensor brands
  - Matching sleeplog and activitylog files.

commit 1d268ab1 (2024-04-16): speeding up create_test_acc_csv fixes #963
PR #1108 (2024-04-16): Speedup create_test_acc_csv (as only used for unit tests)

Fixes #963 by speeding up the create_test_acc_csv.R function. The writing of the file was by far the slowest component and I have now replaced this by `data.table::fwrite`.

No need to mention this in changelog as it does not affect GGIR behaviour for the user.

Checklist before merging:

- [ ] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [x] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [ ] Clean code has been attempted, e.g. intuitive object names and no c [... cut]
closed 2024-04-16 by vincentvanhees