wadpac/GGIR #968: Shell script fails with error message: "'from' must be a finite number and "$call seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) * ws3)), by = ws3)" (Issue #962)

charleslucenap, 2023-11-06:
Hello all,

I've got a database with data from 1014 participants who wore the GENEA Accelerometer.

In GGIR, after part 1, the R program presents the following: "$call seq.default(starttime3, (starttime3 + ((nrow(metashort) - 1) * ws3)), by = ws3)

In addition: Errors and warnings for 011057_01-04-2014 16-16-03.bin$message [1] "'from' must be a finite number

After that, in part 4, the R program generates just 860 files instead of 1014 (original). Probably, I think that these differences are about this "error".

In part 5, I received the following message as well: "Error in 1:max(summarysleep_tmp$night) : "

I am using the GENEActiv, as well as .bin file. Curiously, these data were performed with other cut-offs in the past, and everything works well. However, I'm updating the cut-off and identifying these errors.

I've been looking at other posts that mentioned idloc and colid, however, I tried to modify that but doesn't work. Are there potential solutions? If yes, please, let me know.

Many thanks for this help.

**To Reproduce**
graphics.off() #Ensure that there are no graphic devices going on
rm(list=ls())  #Removing all variables/objects from the working environment
library(GGIR)

##############
#INPUT NEEDED#
##############

datadir = "C:/Pelotas Cohort/Epidemiology/Accelerometer/csv" # directory where accelerometer files are / list of accelerometer files incl. path
studyname = ""#mystudy" #name of sutdy, only needed if datadir is a list of filenames
outputdir = "C:/Results" #name directory where output needs to be stored
overwrite=FALSE

f0 = 1 #file to start with
f1 = length(list.files(datadir)) #file to end with if used in serial analyses (modify accordingly, if infinite then it will process until last file)

###################################
#NO INPUT NEEDED FROM HERE ONWARDS#
###################################

#Function (named my.g.shell.GGIR) that will be used to run modes 1-5
GGIR(#-------------------------------
     # General parameters
     #-------------------------------
     mode=c(1,2,3,4,5), 
     datadir=datadir, 
     outputdir=outputdir, 
     studyname=studyname, 
     f0=f0, 
     f1=f1, 
     desiredtz ="America/Sao_Paulo",
     overwrite = FALSE, 
     do.imp=TRUE,
     idloc=1, 
     print.filename=TRUE,
     storefolderstructure = FALSE,
     #-------------------------------
     # Part 1 parameters:
     #-------------------------------
     windowsizes = c(5,900,3600),
     do.cal=TRUE,
     do.enmo = TRUE,
     do.anglez=TRUE,
     chunksize=1,
     printsummary=TRUE,
     #-------------------------------
     # Part 2 parameters:
     #-------------------------------
     strategy = 1,
     ndayswindow=7,
     hrs.del.start = 0,
     hrs.del.end = 0,
     maxdur = 0,
     includedaycrit = 16,
     L5M5window = c(0,24),
     M5L5res = 10,
     winhr = c(5,10),
     qlevels = c(c(1380/1440),c(1410/1440)),
     qwindow=c(0,24),
     ilevels = c(seq(0,400,by=50),8000),
     mvpathreshold =c(68),
     #-------------------------------
     # Part 3 parameters:
     #-------------------------------
     timethreshold= c(5),
     anglethreshold=c(3),
     ignorenonwear = TRUE,
     #-------------------------------
     # Part 4 parameters:
     #-------------------------------
     #Labels detected sustained inactivity periods by g.part3 as either nocturnal sleep or daytime sustained inactivity
     #Description
     #Loads output from g.part3 as stored in milestone data and sleep log information (if available) and 
     #then uses these information sources to define nocturnal sleep and daytime sustained inactivity.
     
     excludefirstlast = FALSE,
     includenightcrit = 16,
     def.noc.sleep = 1,
     loglocation= c(),
     outliers.only = FALSE,
     relyonsleeplog = FALSE,
     #relyonsleeplog: If TRUE then sleep onset and waking time are defined based on timestamps derived from sleep log 
     #if FALSE (default) the sleep log is only used to guide the accelerometer-based detection. 
     #If participants were instructed NOT to wear the accelerometer during waking hours then set to TRUE, in all other scenarios set to FALSE (FALSE).
     
     colid=1, #Column number in the sleep log spreadsheet in which the participant ID code is stored (default = 1)
     coln1=2, #Column number in the sleep log spreadsheet where the onset of the first night starts
     
     do.visual = TRUE,
     #nnights=5, #Number of nights for which sleep log information should be available.
     #It assumes that this is constant within a study. If sleep log information is missing for certain nights then leave these blank
     sleeplogidnum=TRUE,
     #-------------------------------
     # Part 5 parameters:
     #-------------------------------
     # Key functions: Merging physical activity with sleep analyses
     threshold.lig = c(25),
     threshold.mod = c(68),
     threshold.vig = c(400,428.8,249),
     boutcriter = 0.8,
     boutcriter.in = 0.9,
     boutcriter.lig = 0.8,
     boutcriter.mvpa = 0.8,
     boutdur.in = c(10,20,30),
     boutdur.lig = c(1,5,10),
     boutdur.mvpa = c(1,5,10),
     timewindow = c("WW"),
     #-----------------------------------
     # Report generation
     #-------------------------------
     do.report=c(2,4,5),
     visualreport=TRUE)

#############
#RUN PHASE 1#
#############

#To run in 60 cores in parallel
library(doParallel)
registerDoParallel(60)
my.g.shell.GGIR(mode=1, f0=f0, f1=f1)

###############
#RUN PHASE 2 - 5#
###############
my.g.shell.GGIR(mode=2, f0=f0, f1=f1)
my.g.shell.GGIR(mode=3, f0=f0, f1=f1)
my.g.shell.GGIR(mode=4, f0=f0, f1=f1)
my.g.shell.GGIR(mode=5, f0=f0, f1=f1)

1. Sensor brand: 'GENEA.'
2. Data format: '.BIN'
3. Approximate recording duration 6-7 days
4. Are you using a sleep diary to guide sleep detection: YES
5. Copy of R command used: 'I listed above'
6. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES

**Desktop (please complete the following information):**
 - OS: [e.g. Windows]
 - GGIR Version [e.g. 3.0-0]

charleslucenap, 2023-11-07:
Btw, I tried run again, and the R send me the following message: 
[image]

[image]

vincentvanhees, 2023-11-10:
The errors indicate that first something goes wrong in part 1. So, focus on that and do not attempt to process more parts before the part 1 error is resolved. To do this, create a GGIR call with the minimum number of arguments and mode = 1 and desiredtz ="America/Sao_Paulo". Use this to process a few files. If that works process more files. If you are able to process a large sample with this without error. Try again the same sample but now with different input arguments.

Once you identify the error try to isolate which file is causing it and identify for that file whether you can process it with the default GGIR parameters or whether it only occurs when you change parameters.

Please report back when you have a detailed description of the exact conditions under which the problem occurs. When you are at that stage you can send me a copy of one problematic file + your R script (v.vanhees at accelting.com) and I will have a look.

vincentvanhees, 2023-11-30:
Closing this issue now, feel free to re-open if you keep having problems. In that case please respond to my suggestions above.

closed 2023-11-30 by vincentvanhees