wadpac/GGIR #971: Error in strsplit(x = uFilename[p], split = "[.]csv") :   non-character argument in part6AlignIndividuals.R

rui4833, 2023-11-13:
When I run the mode 6, I got this error:  Error in strsplit(x = uFilename[p], split = "[.]csv") :   non-character argument 

I rerun the part6AlignIndividuals.R using my data, discovered that the uFilename was the name of RData rather than the csv.

In part6AlignIndividuals.R, line 52:  the original code was:   pat1 = unlist(strsplit(x = uFilename[p], split = "[.]csv"))[1]

I wonder if there was a bug in line 52 and how to solve this.

Thank you so much!

vincentvanhees, 2023-11-15:
Hi, I cannot reproduce your error. Below I am including the script I have been using to test this functionality.
1. Can you spot what you did differently?
2.  Are you trying to reprocess a previously processed dataset? If yes, what happens if you  start in a new output folder reprocessing your files?

```
GGIR(mode = 1:6, 
     datadir = datadir, # defined before this call
     idloc = 1, # study specific
     outputdir = outputdir, # defined before this call
     do.report = c(2, 4, 5),
     do.parallel = TRUE,
     overwrite = TRUE,
     printsummary = TRUE,
     desiredtz = "America/Halifax",
     windowsizes = c(5, 60, 3600), # <= note, I use this to increase resolution of lux value
     #=====================
     # Part 2
     #=====================
     strategy = 1,
     hrs.del.start = 0, hrs.del.end = 0,
     maxdur = 100, includedaycrit = 16,
     excludefirstlast = FALSE,
     includenightcrit = 16,
     winhr = c(5,10),
     closedbout = FALSE,
     epochvalues2csv = TRUE,
     #=====================
     # Part 3 + 4
     #=====================
     outliers.only = TRUE,
     criterror = 0,
     do.visual = TRUE,
     #=====================
     # Part 5
     #=====================
     threshold.lig = c(30), threshold.mod = c(100), threshold.vig = c(400),
     boutcriter = 1,
     boutcriter.in = 1, boutcriter.lig = 1,  boutcriter.mvpa = 0.9,
     boutdur.in = c(1,10,30),
     boutdur.lig = c(1,10),
     boutdur.mvpa = c(1),
     
     save_ms5rawlevels = TRUE,
     save_ms5raw_format = "RData",
     save_ms5raw_without_invalid = FALSE,
     part5_agg2_60seconds = TRUE,
     frag.metrics = "all", 
    
     #=====================
     # Part 6
     #=====================
     part6HCA = TRUE,
     part6_threshold_combi = "30_100_400",
     #=====================
     # Visual report
     #=====================
     timewindow = c("WW"),
     visualreport = TRUE)
```

rui4833, 2023-11-15:
Thank you so much~ The problem was solved!
My original code set idloc = 2, which end up with this error.

rui4833, 2023-11-16:
Yes, I was trying to reprocess a previously processed dataset which finished previous 5 parts. It still worked when I set overwrite=F.
I want to know whether the argument closedbout = FALSE was deprecated already?

vincentvanhees, 2023-11-23:
> I want to know whether the argument closedbout = FALSE was deprecated already?

Yes, it has been deprecated

closed 2023-11-23 by vincentvanhees