wadpac/GGIR #975: GGIR Error part 2 day summary

astegman123, 2023-11-14:
Hi! Thank you for making this code. It has been very helpful in analyzing our data so far. 
We have been using the GGIR code to analyze GENEActiwatch data for since the summer and its been working but I recently ran into an issue with the code. It gave the error below after running the GGIR code. I have not changed any of the code and I could not figure out what the error code was referencing. We still had some data output but it looks different than previous times when we have used the code. 

Error message:

Error in if (P2daysummary_tmp$`N hours`[1] < 24 & P2daysummary_tmp$`N hours`[1] >  : 
  missing value where TRUE/FALSE needed

Thank you for your help and taking the time to figure this out!

vincentvanhees, 2023-11-15:
When you open a new issue GitHub provides you with a template, please use that template in order to make it easier for people like me to investigate your problems. In short, I need a reproducible description.

astegman123, 2023-11-17:
Hi! Sorry, I did not realize there was a template. I have included all of the information below. Thank you for your help!

Describe the bug
at the end of the code being run there is an error that pops up saying:
Error in if (P2daysummary_tmp$`N hours`[1] < 24 & P2daysummary_tmp$`N hours`[1] >  : 
  missing value where TRUE/FALSE needed
We have been using our code successfully for a few months and have not changed it prior to receiving this message
To Reproduce
Steps to reproduce the behavior.
Sensor brand: GENEActiwatch
Data format: .BIN
Approximate recording duration: 7 days
Are you using a sleep diary to guide the sleep detection: NO
Copy of R command used:
ONLY RUN THESE TWO LINES IF YOU HAVEN'T PREVIOUSLY INSTALLED: install GGIR and reader for GENEActiv devices ###
install.packages("GGIR", dependencies = TRUE)
install.packages("GGIRread")
Part 1 introduces program and directory, mode indicates which parts to run (should run all), note to use forward slashes in directory ###
library(GGIR)
library(GGIRread)
GGIR(mode=c(1,2,3,4,5),
datadir="C:/Users/u#######/Box/KINES Depner Lab/MASTER LAB Folder/Studies/TOCS/Analyses and Data/Actigraphy/Data",
outputdir="C:/Users/u#######/Box/KINES Depner Lab/MASTER LAB Folder/Studies/TOCS/Analyses and Data/Actigraphy/Output",
do.report=c(2,4,5),
overwrite = TRUE,
windowsizes = c(5,900,3600),
do.cal = TRUE,
do.enmo = TRUE, do.anglez=TRUE,
chunksize=1, printsummary=TRUE,
=====================
Part 2: 
strategy = 1 means select data based on hrs.del.start and hrs.del.end, strategy = 2 makes that only the data between the first midnight and the last midnight is used, 3 is most active days, 4 is only after 1st midnight ###
=====================
strategy = 1,
hrs.del.start = 0, hrs.del.end = 0,
maxdur = 9, includedaycrit = 16,
qwindow=c(0,24),
mvpathreshold =c(100),
excludefirstlast = FALSE,
includenightcrit = 16,
=====================
Part 3 + 4:
I think outliers.only - do.visual only apply if sleep log data present, but keeping in for now ###
=====================
def.noc.sleep = 1,
timethreshold= 5, anglethreshold=5,
ignorenonwear = TRUE,
outliers.only = FALSE,
criterror = 4,
do.visual = TRUE,
=====================
Part 5
=====================
threshold.lig = c(30), threshold.mod = c(100), threshold.vig = c(400),
boutcriter = 0.8, boutcriter.in = 0.9, boutcriter.lig = 0.8,
boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig = c(1,10),
boutdur.mvpa = c(1),
includedaycrit.part5 = 2/3,
=====================
Visual report
=====================
timewindow = c("WW"),
visualreport=TRUE)

Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES 
Expected behavior
I expected to receive a GGIR data output with no error message. 
Screenshots
If applicable, add screenshots to help explain your problem. Note that usually we are not only interested in see the error message in red, but all GGIR output to the console.

[... 98 lines cut ...]
 Report part 2 
 1
____________________________________________________________________________________________________
 Report part 4 
 loading all the milestone data from part 4 this can take a few minutes

report not stored, because no results available
____________________________________________________________________________________________________
 Report part 5 
 loading all the milestone data from part 5 this can take a few minutes
 generating csv report for every parameter configurations...
 WW-30-100-400-T5A5
____________________________________________________________________________________________________
 Generate visual reports 
Error in if (P2daysummary_tmp$`N hours`[1] < 24 & P2daysummary_tmp$`N hours`[1] >  : 
  missing value where TRUE/FALSE needed

Desktop (please complete the following information):
OS: [e.g. iOS]-- Windows 10
GGIR Version [e.g. 2.2-0]: 3.0-0
Additional context
we still have a data output but I'm concerned it is not correct since it is formatted differently than normal
Before submitting
[ ] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there? Yes, Yes, Yes
[ ] Have you tried this on the latest master branch from GitHub? yes

astegman123, 2023-11-20:
UPDATE: this seems to only be an issue with one of the GENEActiwatches

vincentvanhees, 2023-11-23:
Thanks for reporting back, I will now close this issue. Feel free to re-open if the issue comes back.

closed 2023-11-23 by vincentvanhees

vincentvanhees, 2023-12-01:
... actually, if you can still produce a minimally reproducible example of the error then that would be helpful.
For example, the corresponding milestone data file in the meta/basic output folder.