wadpac/GGIR #983: Horizontal axis QC plot not clear

vincentvanhees, 2023-11-27:
@jhmigueles, I just created a QC plot and see:

[image]

which seems to be the result of the changes you made a few months ago in this commit: https://github.com/wadpac/GGIR/commit/e79190f1499d43d6534c5b0b8e700adf2aa0e891

Is this expected behaviour? It is hard now to see how many days there are and also it is unclear why time (12) does not increase. 

Could it be that the horizontal axis label is incorrect and should read "time in each day" as opposed to time in the recording ? But even then I do not think the image is informative this way. I think it would be better to have both day numbers to quickly see what day it is and to replace 12 by the word noon or by midnight because 12am and 12pm are both 12.

jhmigueles, 2023-11-27:
Yes, this is expected but clearly not ideal. I did not try this in a file with many days as you have here (it looked better in a 7-day recording. The ticks without labelling should state "0", but they are not there for limited horizontal space. Would it be ok if I set 2 x axes? One with time in the day (noon/midnight as you suggest), and another one below with the day counting?

vincentvanhees, 2023-11-27:
Yes, sounds good. And then maybe use a slightly smaller font size or reduce the font size if there are more than 10 days of data?

jhmigueles, 2023-11-28:
@vincentvanhees Does this look better? Let me know if you have any extra feedback:

[image]

vincentvanhees, 2023-11-29:
looks good. I am wondering whether it can be more compact and still be clear. How about: "12:00" and "24:00" OR "12:00" and "00:00" OR "12h" and "24h"?

vincentvanhees, 2023-11-29:
Actually. The labels for noon and midnight may not even be needed anymore, because day indicates start and end of the day and the ticks and noon and midnight are therefore automatically clear as such. So what you have is good but just remove the words noon and midnight. Agree?

jhmigueles, 2023-12-01:
Better? I am doing a PR now so that you can test this in your data and give extra feedback if needed:
[image]

PR #985 (2023-12-13): Issue983 xaxis q cplot

Closes #983 => Horizontal axis of QC plot is now redefined to make cleares the days length and also marking midnights and noons.

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [ ] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [ ] Updated or expanded the documentation.
- [x] Updated release notes in `inst/NEWS.Rd` with a user-readable summary. Please, include references to relevant issues or PR discussions.
- [ ] Added your name to the contributors lists in the  [... cut]
closed 2023-12-13 by vincentvanhees