wadpac/GGIR #1002: Error in  rbind(deparse.level, ...) in g.report.part5

Minjingy, 2023-12-18:
**Describe the bug**
When I tried to use GGIR to analysis the ActiGraph.csv data from NHANES, I met a bug and cannot get the PART 5 and PART 6 report summary data, though all the meta data in part1-6 already existed:   
Error in rbind(deparse.level, ...) : 
  numbers of columns of arguments do not match
Calls: GGIR ... g.report.part5 -> as.data.frame -> do.call -> <Anonymous> -> rbind
Execution halted

**To Reproduce**

1. Sensor brand: 'ActiGraph model GT3X+'
2. Data format: 'csv'
3. Approximate recording duration '1-7' days
4. Are you using a sleep diary to guide the sleep detection: NO
5. Copy of R command used: '....'
library(GGIR)
GGIR(
  mode=c(1,2,3,4,5,6),
  datadir="/mbak/wange/20231120_2011-2012acc/batch19",
  outputdir="/mbak/wange/Results_20231120_2011-2012acc/batch19",
  do.report=c(2,4,5,6),overwrite = TRUE, maxNcores=8,
  #=====================
  # read.myacc.csv arguments
  #=====================
  rmc.nrow = Inf,
  rmc.skip = 0,
  rmc.dec = ".",
  rmc.firstrow.acc = 2,
  rmc.col.acc = 2:4,
  rmc.col.time=1,
  rmc.unit.acc = "g",
  rmc.unit.time = "POSIX",
  rmc.format.time = "%Y-%m-%d %H:%M:%OS",
  desiredtz = "America/Chicago",
  rmc.sf = 80, 
  #=====================
  # Part 2
  #=====================
  maxdur = 9, cosinor = TRUE,IVIS.activity.metric = 2,IVIS_acc_threshold = log(20 + 1), IVIS_per_daypair = TRUE,                
  #=====================
  # Part 5
  #=====================
  do.sibreport = TRUE, outliers.only=FALSE, possible_nap_window = c(9,19),
  possible_nap_dur = c(15,240),nap_model = "hip3yr", frag.metrics="all",
  save_ms5rawlevels = TRUE,
  save_ms5raw_format = "RData",
  save_ms5raw_without_invalid = FALSE,
  part5_agg2_60seconds = TRUE,
  #=====================
  # Part 6
  #=====================
  part6CR = TRUE
)
6. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? NO

**Expected behavior**
I hope to have the PART 5 and PART 6 report summary data, especially for "nap_totalduration" and "nap_count" items and fragmentation variables in PART 5 and circadian rhythm variables in PART 6, otherwise I need to write separate codes to summarize these variables for each person.

**Screenshots**
PART 2/4 summary data and meta results were available: 
[image]
[image]
But Report part 5 and 6 cannot be exported:
[image]

**Desktop (please complete the following information):**
 - OS: Linux system in the server
 - GGIR Version: 3.0.1

**Additional context**
Actually, I split the ActiGraph.csv data from NHANES into 20 batches and I successfully runed other batches but this error happened in two of these batches. These data are in the same format for different batches, and I don't know why this error occurs.

**Before submitting**
- [ ] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there? Yes, I reran the codes and the issue still appeared. 

- [ ] Have you tried this on the latest `master` branch from GitHub? Yes, I tried 3.0.2 version GGIR and the results were also the same.

vincentvanhees, 2023-12-18:
It seems GGIR is struggling to append the part 5 output data from two recordings.

To help investigate the issue could you do the following:

1. make a backup copy of the output subfolder `meta/ms5.out` for one of the problematic batches.
2. delete the content of the original `ms5.out` folder
3. now copy a small set of `ms5.out` from your backup into `ms5.out` and run GGIR with `mode =c()`, `do.report = 5`, and `visualreport = FALSE`. If this reproduces the error go to next step, otherwise try again with a different subsample. If you have a lot of files you could also do this in reversed order, first start with all files and delete just as many to keep getting the error.
4. once you have reproduced the error, please send me the small subset of RData files from the `ms5.out` that allowed you to reproduce the error.

Minjingy, 2023-12-19:
Thank you for your swift reply! I successfully solved this problem by following your suggestions. I tried the above method and lastly found 1 file in the ms5.out which had reproduced the error, after excluding this file the remaining 788 files in the meta/ms5.out can be exported to the PART 5 summary report.   I have already sent the RData file from the ms5.out that reproduce the error and the source file to your email v.vanhees@accelting.com.

Minjingy, 2023-12-19:
Actually, this file in the ms5.out which had reproduced the error [(Error in rbind(deparse.level, ...) in g.report.part5)] is the first file in the ms5.out folder, with only 165 columns, and the second file has 191 columns. So, I guess maybe the format of the output data frame is based on the first file so after excluding this first file the remaining files in the meta/ms5.out can be exported to the PART 5 summary report, and the two of 20 batches of my data just had the same problem, but the other 18 batches can run successfully.

vincentvanhees, 2024-01-11:
Thank you for sending the test files. I have prepared a solution. Would you mind testing it by installing this GGIR development version:

`remotes::install_github("wadpac/GGIR", ref = "issue1002_handling_inconsistent_cols_part5")`

To test it you will not need to reprocessed part 5. Just re-run the report generation with arguments: `mode = NULL` and `do.report = 5`.

PR #1017 (2024-01-30): Fixing various smaller issues

This PR addresses a couple of minor issues, each of which I prepared in its own branch, but having one combined PR may be easier:

- Fixes #1002: When part 5 milestone data does not have consistent number of columns this causes an error. My fix adds a warning to inform that user what the cause could be, but also it tries to more intelligently initialise the column names to avoid this issue.
- Fixes #1011: Revise NotWorn algorithm because it relied on the 10th percentile in the distribution of counts which is not informative if the night is full with zeros. 
- Fix [... cut]
commit 86ca33e9 (2024-01-30): fixes #1002
closed 2024-01-30 by vincentvanhees