wadpac/GGIR #1010: Data frame error in Report Part 2

Asanto32, 2024-01-05:
Description
The following error occurs when running the ggir shell on gt3x data

```
 Report part 2 
 1Error in `[.data.frame`(df, 16:ncol(df)) : undefined columns selected
Calls: g.shell.GGIR ... g.convert.part2.long -> which -> apply -> [ -> [.data.frame
Execution halted
```
This occurs when running on a linux server through a docker image (built on amd64), GGIR version: 3.0.2. This only occurs with one specific .gt3x file (so far).

This is the full list of arguments called in ggir:
```
g.shell.GGIR(#=======================================
              # INPUT NEEDED:
              #mode=c(1:5), default is 1:5
              datadir=file_path,
              outputdir="/output",
              studyname=study_path,
              overwrite = TRUE,
              idloc=6,
              print.filename=TRUE,
              storefolderstructure = FALSE,
              dformat=2,
              mon=3,
              do.parallel=FALSE,
              maxdur = 10,
              #-------------------------------
              # Part 1:
              #-------------------------------
              desiredtz = "America/New_York",
              do.enmo = TRUE,
              do.anglez=TRUE,
              do.hfen = FALSE,
              windowsizes = c(10,600,1800),
              # do.cal=TRUE, default is true
              chunksize=1,
              printsummary=TRUE,
              #-------------------------------
              # Part 2:
              #-------------------------------
              strategy = 4,
              hrs.del.start = 0,
              hrs.del.end = 0,
              #maxdur = 0, default is 0
              #includedaycrit = 16, # for midnight-midnight
              winhr = c(5),
              epochvalues2csv=TRUE,
              L5M5window = c(0,24),
              M5L5res = 10,
              qlevels = c(c(1380/1440),c(1410/1440)),
              qwindow=c(0, 12, 24),
              ilevels = c(200,400,800),
              mvpathreshold =c(140),
              do.imp=TRUE,
              #-------------------------------
              # Part 3:
              # (detection of sustained inactivity periods as needed for sleep detection in g.part4)
              #-------------------------------
              timethreshold= c(10),
              anglethreshold = 4,
              ignorenonwear = TRUE,
              nonwear_approach = "2023",
              #-------------------------------
              # Part 4:
              # （Labels detected sustained inactivity periods by g.part3 as either nocturnal sleep or daytime sustained inactivity)
              #-------------------------------
              excludefirstlast = FALSE,
              includenightcrit = 12,
              def.noc.sleep = c(1),
              outliers.only = FALSE,
              relyonguider = FALSE,
              sleeplogidnum = TRUE,
              do.visual = TRUE,
              do.sibreport = TRUE,
              criterror = 0,
              #-------------------------------
              # Part 5:
              #-------------------------------
              threshold.lig = c(30),
              threshold.mod = c(125),
              threshold.vig = c(400),
              boutcriter = 0.8,
              boutcriter.in = 0.9,
              boutcriter.lig = 0.8,
              boutcriter.mvpa = 0.8,
              boutdur.in = c(10,20),
              boutdur.lig = c(1,10),
              boutdur.mvpa = c(1,10),
              timewindow = c( "MM"),
              # -----------------------------------
              # Report generation
              # -----------------------------------
              do.report=c(2,4,5),
              visualreport=TRUE,
              dofirstpage = TRUE,
              viewingwindow=2)
}
```

l-k-, 2024-01-10:
I can take a look at this if no one is working on it? But I'd need access to the .RData file from the meta/basic folder, for the one gt3x  file that caused this.

I think the error is happening on [this line](https://github.com/wadpac/GGIR/blob/master/R/g.convert.part2.long.R#L164), for some reason there are fewer than 16 columns in `df`. But I can't tell what kind of issue with input data would cause this. And it's probably better to figure this out, rather than just patching that one line of code to not crash when there are fewer `df` columns than expected.

But -- maybe others will know right away what's going on there.

vincentvanhees, 2024-01-10:
Based on only the error message my initial thought was that this may relate to function g.report.part2() attempting to append two part2 milestone data which are somehow not compatible. However, that would not be an explanation if the issue occurs even with a single file. 

@Asanto32 could you:
- Confirm that this happens een if you start with an empty output folder and process only that single file?
- Share the .RData file @l-k- refers to with us? (you have our email addresses)

Asanto32, 2024-01-10:
Ok thank you. I'm currently running a big batch job on the server and encountering some erros. I will compile them in a spreadsheet, confer with Nathalia on our team here who has a bit more expertise, and then hopefully send a more complete list of all the errors I have encountered.
Note that I always start with an empty folder (GGIR is run on each file, the script iterates through each file extension based on participant name, and thus each file has its own output_filename/ folder)

l-k-, 2024-02-28:
The same issue was reported in https://groups.google.com/g/rpackageggir/c/PzYNyZzOLyQ

PR #1070 (2024-03-01): Issue 1010 in g.convert.part2.long(), don't evaluate columns that aren't there

Fixes #1010

When data has no days with enough valid hours, g.analyse.perday() [doesn't add per-day-segment features to daysummary](https://github.com/wadpac/GGIR/blob/75f67093d49e440da08fc26999603576d4b17716/R/g.analyse.perday.R#L273)

Later in g.convert.part2.long(), we were attempting to [validate these features](https://github.com/wadpac/GGIR/blob/75f67093d49e440da08fc26999603576d4b17716/R/g.convert.part2.long.R#L164) and this caused the method to crash.

To fix this, I changed the code in g.convert.part2.long( [... cut]
closed 2024-03-01 by vincentvanhees