wadpac/GGIR #1036: Improve documentation on ActiGraph csv files

vincentvanhees, 2024-02-05:
In the vignettes we refer to Actigraph csv without providing much detail as if it is supposed to be obvious what we mean:
https://cran.r-project.org/web/packages/GGIR/vignettes/readmyacccsv.html
https://cran.r-project.org/web/packages/GGIR/vignettes/GGIR.html

To do:
- [ ] Add clarification what format GGIR expects by referring to the test file GGIR can create as example.
- [ ] Clarify distinction between actigraph count csv files and actigraph raw csv files, and point reader to documentation for processing count csv files.

dsjoho, 2024-12-10:
This would be clarifying! Also for when one has a non actigraph file in order to determine what those files look like.

I'd also like to propose a clarification regarding https://cran.r-project.org/web/packages/GGIR/vignettes/readmyacccsv.html
As a new user, it is unclear to me if I can load an entire directory with the proposed `read.myacc.csv` solution, or if I'm then forced to only read single files. (This is probably obvious after a while, but initial reading is unclear)

Shengkui-Zhang, 2025-03-17:
Subject: Error in GGIR Part 1: "**No files to analyse**" when using read.myacc.csv with CSV file
Dear Vincent ,

I am encountering an issue when trying to run GGIR with a CSV file using the read.myacc.csv function. Here are the details of my problem:

Steps to Reproduce
Reading CSV file with read.myacc.csv:
I used the example data and code provided in the [GGIR vignette](https://cran.r-project.org/web/packages/GGIR/vignettes/readmyacccsv.html) to read a CSV file. This part works perfectly:

R code：
library(GGIR)
data = read.myacc.csv(
  rmc.file = "C:/mystudy/mydata/datafile.csv",
  rmc.nrow = Inf,
  rmc.skip = 0,
  rmc.dec = ".",
  rmc.firstrow.acc = 2,
  rmc.col.acc = 2:4,
  rmc.col.temp = 5,
  rmc.col.time = 1,
  rmc.unit.acc = "g",
  rmc.unit.temp = "C",
  rmc.unit.time = "POSIX",
  rmc.format.time = "%d/%m/%Y %H:%M:%OS",
  desiredtz = "Europe/London",
  rmc.sf = 100
)
The data is read successfully, and I can inspect the contents of data.

When I try to run GGIR with the same CSV file, I get the following error:

R code：
library(GGIR)
GGIR(
  mode = c(1, 2, 3, 4, 5),
  datadir = "C:/mystudy/mydata/datafile.csv",
  outputdir = "D:/myresults",
  do.report = c(2, 4, 5),
  #=====================
  # read.myacc.csv arguments
  #=====================
  rmc.nrow = Inf, 
  rmc.dec = ".",
  rmc.firstrow.acc = 2, 
  rmc.col.acc = 2:4, 
  rmc.col.temp = 5, 
  rmc.col.time = 1,
  rmc.unit.acc = "g", 
  rmc.unit.temp = "C", 
  rmc.unit.time = "POSIX",
  rmc.format.time = "%d/%m/%Y %H:%M:%OS",
  desiredtz = "Europe/London",
  rmc.sf = 100,
  rmc.noise = 0.013
)
The error message is:
Error in g.part1(datadir = datadir, metadatadir = metadatadir, f0 = f0,  : 
No files to analyse. Check that there are accelerometer files in the directory specified with argument datadir
What I Have Tried
Confirmed file existence and path:
The file C:/mystudy/mydata/datafile.csv exists, and the path is correct. This is confirmed by the fact that read.myacc.csv can read the file successfully.

Checked datadir parameter:
I tried pointing datadir to both the file itself (C:/mystudy/mydata/datafile.csv) and its parent directory (C:/mystudy/mydata/). Neither worked.

Tried different GGIR versions:
I tested both GGIR 3.2.1 and 3.0.0, but the issue persists.

Enabled verbose logging:
I added verbose = TRUE to the GGIR call, but no additional information was provided.

Questions
Is GGIR supposed to work with CSV files directly when using read.myacc.csv? If so, what could be causing this issue?

Are there any additional parameters or configurations I need to set to make GGIR recognize the CSV file?

Could this be a bug in how GGIR handles datadir when using read.myacc.csv?

Additional Information
GGIR version: 3.2.1 (also tested with 3.0.0)

R version: 4.3.1

Operating system: Windows 11

Sample data: The example CSV file from the [GGIR vignette](https://cran.r-project.org/web/packages/GGIR/vignettes/readmyacccsv.html).

Thank you for your time and assistance! I look forward to your guidance on resolving this issue.

Best regards

vincentvanhees, 2025-03-17:
@Shengkui-Zhang please do not post an issue as a response to another issue. That is bad practise as it creates confusion about which of the multiple issues is being discussed and solved. Just start a new issue for a new problem or question.