wadpac/GGIR #1042: Add check that windowsizes[3] should be multitude of windowsizes[2]

vincentvanhees, 2024-02-12:
By default `windowsizes = c(5, 900, 3600)`, but when sets to `c(5, 900, 1200)` it does not make sense as the 20 minute window will overlap with 1.333 15 minute window, by which the labelling will be ambiguous. Therefore, I think we need to add to check_params.R:

```
if (ws2/ws != round(ws2/ws)) {
  ws = ws2 * ceiling(ws/ws2)
  warning(paste0("The third value of parameter windowsizes needs to be a multitude of the second value.\n",
                    "The third value has been automatically adjusted to ",
                     ws, " seconds in order to meet this criteria.\n" call. = FALSE)
}
```
as a last step to the check of the windowsizes values.

This has only become relevant with setting `nonwear_approach = "2023"` where windows are expected to allign, which is why I did not account for it previously.

The relevant parts of the code are brought together in check_params in PR #1027 in https://github.com/wadpac/GGIR/blob/part1-simplification-v2/R/check_params.R.  So, it is probably easier to do the above when that PR is merged or to add it as an extra commit to that branch.

l-k-, 2024-02-14:
@vincentvanhees I added the suggested check as a [commit](https://github.com/wadpac/GGIR/pull/1027/commits/a558678a572574b1357e20517a36bfe33684cd25) in PR #1027

l-k-, 2024-02-17:
Completed in #1027

l-k-, 2024-02-17:
Completed in https://github.com/wadpac/GGIR/pull/1027

closed 2024-02-17 by l-k-