wadpac/GGIR #1043: Add option to detect sleep period time in part 5 based on nonwear

jhmigueles, 2024-02-12:
When a device has not been worn during a night, then GGIR struggles to define the sleep period time. This is not a problem in part 4 because we expect that night to be excluded from the sleep analyses anyway. But it might be problematic in part 5. Here an example:

[image]

- Sleep period time was not defined in part 4
- Then, part 5 tries to fix this night in order to use the data during the day window, but the sleep period time defined is too short.
- This makes that a very long nonwear window is classified within the "day" segment, making that either this day is excluded from analyses OR, if it still meets the `includedaycrit.part5` setting, a huge overestimation of the sedentary time in that window.

**Proposed solution**
Identify these scenarios in g.part5.fixmissingnight, label the nonwear time as SIB to try to find a more realistic sleep period time for part 5 analyses

jhmigueles, 2024-02-15:
**Update after discussion via email**

This does not happen when SPTs are not defined in part 4, but when the definition is is not so reliable because of high nonwear time in the noon-to-noon (or 6pm-to-6pm) window. To facilitate access to the nonwear time series, it would be easier to handle this in parts 3 and 4. 

I updated the title of the issue and I will start a new branch to work on this.

jhmigueles, 2024-02-20:
I'd like to know what you think of the approach I am investigating before working more on this, I am trying to use invalid time to assist the definition of the SPT in those nights in which there are invalid chunks, as in the example above.

**Current approach**
In the `HASPT` function, if `HASPT.ignore.invalid = FALSE` (default), then GGIR defines `nomov` vector using the imputed angle. I think it would make more sense to consider invalid time as nomov, as the imputation of the angle provides the means of the rest of the days that could (or could not) be a good indicator of the nomov periods in the night of interest.

**New approach**
What I did is to consider that invalid time as no movement, so invalid periods are turned 1 in the nomov vector if `HASPT.ignore.invalid = FALSE`. This seems to work well, see the example in the first comment using this new approach (I have tested this in more nights and could not find any unexpected classification). Below how this example look like with and without sleeplog used.

- Without sleeplog:
[image]

- With sleeplog (it trusts the sleeplog times in the SPTs with invalid time, in this case, 7am for the wakeup):
[image]

vincentvanhees, 2024-02-21:
In your new approach do you mean something like:

```
invalid = adjustlength(x, invalid)
if (HASPT.ignore.invalid == TRUE) {
  nomov[which(x < pp & invalid == 0)] = 1
} else {
  nomov[which(x < pp | invalid == 1)] = 1
}
```

Ok, I think need facilitate the following scenarios:
1. 24/7 wear protocols where occasionally the participant takes off the sensor at night, even though that was not expected. Here, we want those nights to be classified as SPT.
2. 24/7 wear protocols where we actually want to rely on the imputation to help ignore brief non-wear periods before going to bed or after waking up, and include long nonwear periods during the middle of the night.
3. night-only protocols where the accelerometer is taken off during the day?
4. day-only protocols where the accelerometer is taken off at night.

When matching these scenario with the code variants we have:
- `x < pp & invalid == 0` is good when we want to ignore all non-wear as in scenario 3.
- `x < pp | invalid == 1` is good when we want to allow 'nonwear' to contribute to sleep regardless of the time of the day it happens in scenario 1 and 4
- `x < pp` is possibly good when we want short (daytime) non-wear periods to be imputed by movement in order not to contribute to SPT, while longer non-wear periods during the night to be imputed as sleep to contribute to SPT. This is relevant for scenario 1 and 2.

This may no longer be a binary decision then, and maybe sleep also needs its own `data_masking_strategy`.

vincentvanhees, 2024-02-28:
Closing this issue now. Writing "closes issue #1043" does not close the issue. Only when we type closes #1043 or fixes #1043 the issue will actually be closed.

closed 2024-02-28 by vincentvanhees