wadpac/GGIR #1090: Minor bug in g.conv.actlog when reading dates

jhmigueles, 2024-03-15:
When reading dates from an activity log in `g.conv.actlog`, the function `data.table::fread` loads the date columns with `class = "IDate" "Date" `. This triggers an error in check_log as it expects `class = character`.

Propose including this line in `g.conv.actlog` to make sure that all columns are introduced as character in `check_log`:
`actlog[, -1] = apply(actlog[, -1], 2, as.character)`

vincentvanhees, 2024-03-15:
It would be good to understand what makes your dates different, because the function has worked for many years.  When I load a test csv file with a date column, fread just loads it as character.

The risk of first allowing `fread` to interpret the column as dates is that fread may change the format which then longer matches the format as provided by the user. So, if the problem is `fread` then I propose forcing `fread` to read the column as character during the reading process by adding argument `colClasses = "character"`.

jhmigueles, 2024-03-26:
Thanks @vincentvanhees 

I found the source of the problem, `data.table::fread` would only read the date as class `IDate` if the date format is `%Y-%m-%d`. Here an example using the same log with different date formats, see the column classes for the date columns.

## %Y-%m-%d
```
Ymd = data.table::fread("logs/schoolTimes_GGIR.csv")
head(Ymd)
```
[image]

## %d/%m/%Y
```
dmY= data.table::fread("logs/schoolTimes_GGIR2.csv")
head(dmY)
```
[image]

Are you ok about adding the argument `colClasses = "character"` to make sure all columns are assigned the class character?

vincentvanhees, 2024-03-26:
Yes, sounds good.

PR #1094 (2024-04-02): Issues 1090 1093 activity log

Fixes #1090  => activity log dates are now read as character independently of the format of the date.
Fixes #1093  => activity log times are now interpreted considering the epoch size of the analysis to ensure matching of activity log times with time series times in part 5.

Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [x] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Clean code has been attempted, e.g. intuitive object names and no code redundancy. [... cut]
closed 2024-04-02 by vincentvanhees