wadpac/GGIR #1091: New standard csv file to be processed for GGIR

pinweichen, 2024-03-15:
**Is your feature request related to a problem? Please describe.**
No, this is not related to a problem. A new wrist-worn Sony wearable device is used for a research project. I want to create a csv file format that can be device-neutral for all future wearable data. I called this the standard data format. The standard data is saved as a CSV file. I have a simple extra script that cleans any device data into this standardized format. The standard data with the first row as a header and the rest of the data following four columns (timestamp, x, y, z). 

**Describe the solution you'd like**

I would like to collaborate to add this standard file type to the GGIR part 1. I noticed there is a setting for MONITOR$INDEPENDENCE and FORMAT$CSV. I have taken the initiative to modify but having issues in troubleshooting. I have two questions regarding this: 
(1) I have trouble finding where the MONITOR and FORMAT (e.g., MONITOR$GENEActiv or FORMAT$BIN) are set. Would you mind pointing me to where I can modify these two files for part 1 to use? 
(2) GGIR scripts are nested scripts, and RStudio has issues debugging or pausing when errors occur with nested scripts. Do you have a way for me to look into the errors? I particularly have trouble troubleshooting the g.getmeta script. 

Thank you very much.

Benny

l-k-, 2024-03-17:
Benny,

GGIR already supports a device-neutral ad-hoc csv file format, you should just use that, no need to add anything.

To make sure the ad-hoc csv format is used, you have to specify the `rmc.firstrow.acc` argument. In your case, you will need to specify `rmc.firstrow.acc = 1` (GGIR will ignore your 1-row column header).

You also need to specify which column has timestamps in it (using `rmc.col.time` parameter; in your case it will be `rmc.col.time = 1`), and which columns have acceleration (using `rmc.col.acc` parameter; in your case it will be `rmc.col.acc = 2:4`). 

You also need to tell GGIR what format your timestamps are in, using `rmc.unit.time` parameter. I'm assuming you are using Unix timestamps. If so, you will need to specify `rmc.unit.time = "UNIXsec"`

Another thing that you need to specify is the sampling rate (frequency) of your data, using `rmc.sf`

You can read more about the ad-hoc csv format in GGIR documentation [here](https://cran.r-project.org/web/packages/GGIR/vignettes/readmyacccsv.html), and you can see an example of a GGIR() call [here](https://cran.r-project.org/web/packages/GGIR/vignettes/readmyacccsv.html#31_Example_using_the_shell_function).

vincentvanhees, 2024-03-18:
@pinweichen In relation to your second point:

> (2) GGIR scripts are nested scripts, and RStudio has issues debugging or pausing when errors occur with nested scripts. Do you have a way for me to look into the errors? I particularly have trouble troubleshooting the g.getmeta script.

GGIR is not a collection of R scripts, but a collection of functions that depend on each other. This is how R packages and software in general works, you need to test them together.

What I often do is:
- Clone the repo from GitHub
- Create new project in RStudio with the GGIR folder as project root.
- Either do `devtools::load(".")` to load all functions or
```
fns = dir("D:/Code/GGIR/R", full.names = TRUE)
fns = fns[grep(pattern = "sysdata", x = fns, invert = TRUE)]
for (i in fns) source(i)
```
The second option has the advantage that it also works when you are not in the same project.
- Make sure you do not do `library(GGIR)`.
- Use `browser()` inside the code where relevant to inspect what is happening inside.
- Run GGIR with `do.parallel=FALSE` to ease debugging, the code will then stop when it sees `browser()` or errors if the error happens before `browser`.

vincentvanhees, 2024-03-18:
Will close this issue now as it seems to have been addressed. 
If you cannot get it to work, please open a new issue with a detailed breakdown of what you tried to get it to work including a detailed description of the csv file format or example file.

closed 2024-03-18 by vincentvanhees