wadpac/GGIR #1128: Extracting non-wear from `M$metalong$nonwearscore`?

tgbrooks, 2024-05-16:
I need access to non-wear calling at the epoch level, for downstream analysis. From a previous [discussion](https://github.com/wadpac/GGIR/issues/51), it looks like the way to obtain this is through the Rdata files in the meta directory. However, I was unclear on the specific instructions from that discussion, which refers to the `rout` object. I'm not sure if things have changed since that discussion. I see only `IMP$rout` in the `ms2.out/*.Rdata` file which has columns `r1` through `r5`. I don't see any documentation of these columns - please point me to the right place to look if it exists.

Instead, I found that `data/*.Rdata` contains `M$metalong$nonwearscore`. Is this the right place to get that information? It is coded as as 0, 1, 2, or 3. Am I correct to assume that 0 is wear time and that 1, 2, and 3 should all be treated as non-wear?

For reference, my config is:

```
GGIR(
     datadir = filenames,
     outputdir = outdir,
     studyname = "XYZ",
     idloc = 5,
     # PART 2
     strategy = 2, # use first midnight to last midnight
     desiredtz = "America/New_York",
     includedaycrit = 16, # 16  hours per day needed to include it
     includenightcrit = 16, # 16  hours per day (noon-noon) needed to include it
     # PART 3 + 4
     do.visual=TRUE,
     # OUTPUT:
     epochvalues2csv=TRUE,
)
```

Thank you!

marcellosicbaldi, 2024-05-20:
Hi, 

I am a fellow GGIR user, I can share my experience hoping that it can be useful. As far as I understand, GGIR considers 0 and 1 as wear, while 2 and 3 as nonwear. I got this because whenever I got 1 in the nonwear score, that period was not ignored in the following computations (GGIR ignores nonwear, parameter `ignorenonwear = TRUE` by default). You can also check this from the pdf file `results/QC/plots_to_check_data_quality.pdf`. 

Indeed, in the attached example, you can see that if nonwear_score = 1 (subject B), that period is still labelled as wear by GGIR. If nonwear_score = 2 or 3, then GGIR labels it as nonwear (subject A).

[image]

vincentvanhees, 2024-05-21:
Object `M$metalong$nonwearscore` is an intermediate stage of the non-wear detection as done by GGIR part 1.
GGIR part 2 performans the next step which is then stored in object IMP$rout as documented in the documentation for GGIR function `g.impute`.

Non-wear classifications per epoch are included in the time series as exported with parameters `save_ms5rawlevels` and optionally `save_ms5raw_format` `save_ms5raw_without_invalid` as documented here: https://cran.r-project.org/web/packages/GGIR/vignettes/GGIRParameters.html

For documentation on non-wear detection algorithm in general see:
https://wadpac.github.io/GGIR/articles/chapter3_QualityAssessment.html#non-wear-detection

@marcellosicbaldi `ignorenonwear` is unrelated to the nonwear detection, it relates to how sleep analysis should deal with detected nonwear.

tgbrooks, 2024-05-21:
Thanks for the replies. FYI, I still can't find the documentation of `rout`, or indeed, of `g.impute` in the online documentation: https://cran.r-project.org/web/packages/GGIR/GGIR.pdf It mentions `g.impute` but doesn't appear to have any documentation of it (links to it return me to the top of the pdf). Perhaps something went wrong with the documentation? Or else I'm looking in the wrong PDF.

I did find the documentation by running `?g.impute` in R, with the relevant part below for reference for anyone who finds this:

```
metashort: imputed short epoch variables

    rout: matrix to clarify when data was imputed for each long epoch
          time window and the reason for imputation. Value = 1
          indicates imputation.  Columns 1 = monitor non wear, column 2
          = clipping, column 3 = additional nonwear, column 4 =
          protocol based exclusion and column5 = sum of column 1,2,3
          and 4.

averageday: matrix with n columns for n metrics values and m rows for m
          short epoch time windows in an average 24 hours period
```

closed 2024-05-21 by tgbrooks

vincentvanhees, 2024-05-21:
Yes that is correct. I listed this and many other GGIR functions as internal function such that documentation is only accessible via `?g.impute` and via man folder in the repo, but not via the pdf.

The reason for this is that I received complaints that the pdf was impossible to navigate with more than 100 pages of function documentation including all the internal functions a typical GGIR user should not have to worry about. Actually, the user should only worry about function `GGIR()`.

Further, knowing how this function works is only relevant for someone interested in contributing to GGIR, for which I think it is reasonable to expect familiarity with the `man`  folder. 

I think a better approach to meet your need is to document the content of all the milestone files, which is currently an open issue https://github.com/wadpac/GGIR/issues/752