wadpac/GGIR #1133: Change log issues

vincentvanhees, 2024-05-28:
1. [Change log](https://cran.r-project.org/web/packages/GGIR/news/news.html) does not have release dates which makes it harder to navigate when a change happened:
https://cran.r-project.org/web/packages/GGIR/news/news.html

The dates were included previously but were removed when we integrated pgkdown for gh-pages: https://github.com/wadpac/GGIR/commit/4f2ba569b8aee01c044b40ce9a49071a40d53013

@jhmigueles do you know whether this was really necessary? Ideally the changelog should make it easy to understand what changes happened in a certain time interval.

2. [Changelog on CRAN](https://cran.r-project.org/web/packages/GGIR/news/news.html) is center-aligned on CRAN instead of left-aligned. Why is that?

jhmigueles, 2024-05-28:
Regarding the dates, I remember we discussed this but maybe via email and I cannot find the conversation now. The change to the current `NEWS.md` file was done automatically and the function did not consider the dates. I asked you about whether I should add them manually and you thought that maybe it was not necessary because can always see the dates of the commits.

vincentvanhees, 2024-05-29:
Sorry, I forgot about that. I now realise that it can be helpful to have dates because we refer GGIR users to the changelog for a detailed overview of the changes.

It sounds like adding dates manually will be overwritten when we re-run the pkgdown function. However, I now see that it has the option to include the release dates: https://pkgdown.r-lib.org/reference/build_news.html. So, it may be good to experiment with that to see how different configurations of the _pkgdown.yml file affects NEWS.md.

Regarding alignment, I now see this is also done for other packages: https://cran.r-project.org/web/packages/ggplot2/news/news.html  so I guess it is a change in CRAN.

commit 90ee7f5c (2024-06-04): List cran release dates in the changelog in gh-pages, fixes #1133
closed 2024-06-04 by vincentvanhees