wadpac/GGIR #1174: part3 plot error: seq.int(0, to0 - from, by) : 'to' must be a finite number

vincentvanhees, 2024-07-15:
In a recent [google group post](https://groups.google.com/g/RpackageGGIR/c/M14Tfmxczuc/m/xmc7oeHyAgAJ) the following error was reported:

`seq.int(0, to0 - from, by) : 'to' must be a finite number`

This occurs at the plot generation in GGIR part 3 and for a timezone range in a specific timezone and base function `plot` not being able to handle this. 

I have posted it to stackoverflow: https://stackoverflow.com/questions/78751084/plot-error-seq-int0-to0-from-by-to-must-be-a-finite-number

It would be good to understand what the actual problem to make sure we address this well.
However, if no fix can be found quick work arounds are:
- It is specific to `do.part3.pdf = TRUE`, so setting this to FALSE would be a quick work around.
- Another work around seems to wrap `as.numeric()` around `time[qqq1:qqq2]` line 84 of `g.sib.plot`.

vincentvanhees, 2024-11-06:
I have added the "help wanted" label as I do not have time and am not paid to work on this at the moment.

fdrincon, 2024-12-23:
I am not sure if this is useful, but the main issue is that in Chile on 2023-Sep-03 there is daylight saving time. So, POSIXct is not working correctly with tz ="Chile/Continental". 

- One possible work around would be to use tz ="UTC". 

Rstudio, 
R version 4.4.2 (2024-10-31)
Platform: x86_64-apple-darwin20
Running under: macOS Big Sur 11.3