wadpac/GGIR #1187: Sleep Log Intergration under GGIR Version 3.1.2

michelle-kubicki, 2024-08-15:
Hello, I had another issue with sleep log integration. The sleep logs are working great under GGIR version 3.0.6 but we have a lab member with version 3.1.2 and they are not getting the same correct output. 

We are using logs with some days of missing data and format them by just leaving blank spaces for onset and wake under the typical column labels, as instructed on the vignette page. Our code is set to using actigraphy data when sleep log data is unavailable. 

The issue we're running into with the **newer version of GGIR only** is that in the ‘part4_nightsummary_sleep_full’ file in the ‘QC’ folder within the ‘results’ we are looking at, the program, instead of just using the actigraphy data for nights of missing data, uses the next night's sleep log entry. For example, in one file we have missing sleep and wake times for night 7 (out of 15 nights), and instead of using the actigraphy data for that night and showing cleaningcode = 1 on the results sheet, it uses the sleep and wake sleep log entries for night 8. This means that for a sleep log with three days of missing data, the last three days are treated as if they don't have sleep log data (although they do) because the program is just shifting the log entries over. This is not occurring in the older version, the program just default to accelerometer data instead of shifting the log dates.

The code that we are using is below:
---
ONLY RUN THESE TWO LINES IF YOU HAVEN'T PREVIOUSLY INSTALLED: install GGIR and reader for GENEActiv devices ###
install.packages("GGIR", dependencies = TRUE)
install.packages("GGIRread")
Part 1 introduces program and directory, mode indicates which parts to run (should run all), note to use forward slashes in directory ###
library(GGIR)
library(GGIRread)
GGIR(mode=c(1,2,3,4,5),
datadir="C:/Users/u1370752/Box/KINES Depner Lab/MASTER LAB Folder/Studies/TOCS/Analyses and Data/Actigraphy/Data",
outputdir="C:/Users/u1370752/Box/KINES Depner Lab/MASTER LAB Folder/Studies/TOCS/Analyses and Data/Actigraphy/Output",
do.report=c(2,4,5),
overwrite = TRUE,
windowsizes = c(5,900,3600),
do.cal = TRUE,
do.enmo = TRUE, do.anglez=TRUE,
chunksize=1, printsummary=TRUE,
idloc=1,
#=====================
Part 2: 
strategy = 1 means select data based on hrs.del.start and hrs.del.end, strategy = 2 makes that only the data between the first midnight and the last midnight is used, 3 is most active days, 4 is only after 1st midnight ###
#=====================
strategy = 1,
hrs.del.start = 0, hrs.del.end = 0,
maxdur = 16, includedaycrit = 16,
qwindow=c(0,24),
mvpathreshold =c(100),
excludefirstlast = FALSE,
includenightcrit = 16,
#=====================
Part 3 + 4:
I think outliers.only - do.visual only apply if sleep log data present, but keeping in for now ###
#=====================
loglocation="C:/Users/u1370752/Box/KINES Depner Lab/MASTER LAB Folder/Studies/TOCS/Analyses and Data/Actigraphy/Sleep Log/GGIR Sleep Log.csv",colid=1,coln1=2, sleeplogsep=",", desiredtz="",
def.noc.sleep = 1,
timethreshold= 5, anglethreshold=5,
ignorenonwear = TRUE,
outliers.only = FALSE,
criterror = 3,
do.visual = TRUE,
#=====================
Part 5
#=====================
threshold.lig = c(30), threshold.mod = c(100), threshold.vig = c(400),
boutcriter = 0.8, boutcriter.in = 0.9, boutcriter.lig = 0.8,
boutcriter.mvpa = 0.8, boutdur.in = c(1,10,30), boutdur.lig = c(1,10),
boutdur.mvpa = c(1),
includedaycrit.part5 = 2/3,
save_ms5rawlevels = TRUE,
save_ms5raw_format = "csv",
save_ms5raw_without_invalid = FALSE,
#=====================
Visual report
#=====================
timewindow = c("WW"),
visualreport=TRUE)

vincentvanhees, 2024-08-22:
I did identify an issue #1188 that sounds similar to what you describe.
I have now prepared a fix for that issue, would you mind trying it out?

To do so, please install GGIR with:
`remotes::install_github("wadpac/GGIR", ref = "issue_1138_1188")`

this does not have a new version number yet, so ignore the version number.

michelle-kubicki, 2024-08-22:
Vincent, it worked! Thank you so much for working on this.

vincentvanhees, 2024-08-23:
thanks for confirming it works, this will be in the next release.

closed 2024-08-23 by vincentvanhees