wadpac/GGIR #1206: g.part5.wakesleepwindow classifying night from sleeplog might go beyond recording length

jhmigueles, 2024-09-27:
**Describe the bug**
If a night is not detected in part 4, but there is a sleeplog available, g.part5.wakesleepwindows would classify this SPT within the ts using the sleeplog sleeponset and wakeup times. The issue comes when the sleeplog refers to a time which is beyond the recording length. Then, GGIR struggles to assign the SPT indices ts and triggers an error. 

**To Reproduce**
Steps to reproduce the behavior.

1. Sensor brand: any
2. Data format: any
3. Approximate recording duration >= 2 days as this occurs in part 5
4. Are you using a sleep diary to guide sleep detection: YES

**Expected behavior**
If sleeplog refers to a time beyond the recording length, then all the time from the last sleeponset to the end of the recording should be classified as SPT.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - GGIR Version: 3.1-4

**Before submitting**
- [ ] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there?
- [ ] Have you tried this on the latest `master` branch from GitHub?

vincentvanhees, 2024-10-04:
I am struggling to reproduce the error. My test file ends at 13:00, so I set the matching sleeplog wakup time to 14:00. Next I used GGIR parameters to skip the last night:
- `data_masking_strategy = 2`
- `data_masking_strategy = 1` and `hrs.del.end = 20`
- `excludelast.part4 = TRUE`
but for all of them part 5 runs without errors.

I am using the advanced sleeplog format, is this specific to basic sleeplog format?

jhmigueles, 2024-10-07:
I am using advanced sleeplog as well. Here more details on the specific cases I have found:

**1- Last night in sleeplog completely out of recording** 
Recording ending at 0:45, and advanced sleeplog for that day had sleeponset at 1:30 and wake-up at 7:30.
Here, GGIR does not trigger any error, the full time series is classified as sleep period time, and then this participant does not appear in part 5 reports.

**2- Last night in sleeplog partially out of recording and high nonwear** 
Recording ending at 1am, and advanced sleeplog for that day had sleeponset at 22:40 and wake-up at 8:15. In this night, there is a high non-wear in the last night, triggering `g.part5.wakesleepwindows` lines 99 onwards and re-derives sleep onset and wake up for that night (even though it is available in the part 4 output. 
Then, in this case, if the wake-up derived occurs after the recording, GGIR triggers an error.

I can send test files and sleeplogs to replicate these 2 issues.

jhmigueles, 2024-10-07:
Also, I am mostly using default values, except for part 3 and 4, where I am using the sleeplog and, additionally:

- `HASPT.algo = c("NotWorn", "HDCZA")`
- `HASPT.ignore.invalid = NA`
- `ignorenonwear = FALSE`

PR #1207 (2024-10-23): Issue1206 spt part5

Fixes #1206 => g.part5.wakesleepwindows now adjusts the sleeplog times to the recording length in the case that the sleeplog times goes beyond the time series (e.g., recording ending at 6am and sleeplog wakeup time is 8am).

### Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [ ] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Clean code has been attempted, e.g. intuitive object names and no code redundancy.
- [ ] Documentation updated:
  - [ ] Function documentat [... cut]