wadpac/GGIR #1224: Can not process part one, seems to be related to "cannot open the connection" from gzfile(file, "wb")

Chinate, 2024-11-12:
**Describe the bug**
When I try to run GGIR function for my data, an error reported from part 3 that "No milestone data found for part(s) 2.", but the problem seems to start from part 1 since "g.part1" reports "[1] "cannot open the connection" $call gzfile(file, "wb")"

**To Reproduce**
1. create mydata.csv using the same data as the example presented at the end of "https://cran.r-project.org/web/packages/GGIR/vignettes/readmyacccsv.html".
2. place it in directory "/Users/zhiyuan/Desktop/2024_8_26/Project/5/test", and create directory "/Users/zhiyuan/Desktop/2024_8_26/Project/5/out".
3. run `ggir_test <- read.myacc.csv(rmc.file = "/Users/zhiyuan/Desktop/2024_8_26/Project/5/test/mydata.csv",
rmc.nrow = Inf,
rmc.skip = 0,
rmc.dec = ".",
rmc.firstrow.acc = 2,
rmc.col.acc = 2:4,
rmc.col.temp = 5,
rmc.col.time=1,
rmc.unit.acc = "g",
rmc.unit.temp = "C",
rmc.unit.time = "POSIX",
rmc.format.time = "%d/%m/%Y %H:%M:%OS",
desiredtz = "Europe/London",
rmc.sf = 100)`, got a list as expected, which contains a data frame with 9 obs with 5 variables.
4. run `g.part1(datadir = "/Users/zhiyuan/Desktop/2024_8_26/Project/5/test",
metadatadir = "/Users/zhiyuan/Desktop/2024_8_26/Project/5/out",
minimumFileSizeMB = 0,
f1 = 1,
rmc.nrow = Inf,
rmc.skip = 0,
rmc.dec = ".",
rmc.firstrow.acc = 2,
rmc.col.acc = 2:4,
rmc.col.temp = 5,
rmc.col.time=1,
rmc.unit.acc = "g",
rmc.unit.temp = "C",
rmc.unit.time = "POSIX",
rmc.format.time = "%d/%m/%Y %H:%M:%OS",
desiredtz = "Europe/London",
rmc.sf = 100)`, got output that says
"Error in gzfile(file, "wb") : cannot open the connection
In addition: Warning message:
In gzfile(file, "wb") :
  cannot open compressed file 'out//meta/basic/meta_mydata.csv.RData', probable reason 'No such file or directory'"

1. Sensor brand: '...'
4. Data format: 'csv'
5. Approximate recording duration '...' days
6. Are you using a sleep diary to guide the sleep detection: YES / NO
7. Copy of R command used: '....'
8. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES

**Expected behavior**
g.part1 is expected to run without this error message

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS 15.0.1 (24A348) // also tested on windows 
 - GGIR Version 3.1-4 // 3.1-5 on windows

**Additional context**
Also tested with data from https://ftp.cdc.gov/pub/pax_g/ data which contains accelerometer raw data for 7 consecutive days, problem still exists

**Before submitting**
- [ Y] Have you tried the steps to reproduce? Do they include all relevant data and configuration? Does the issue you report still appear there?
- [ ] Have you tried this on the latest `master` branch from GitHub? Tried 3.1-4, 3.1-5

tarensanders, 2024-11-13:
I'm going to jump in to give Vincent a break.

The clue (as always) is in the error message:
> cannot open compressed file 'out//meta/basic/meta_mydata.csv.RData', probable reason 'No such file or directory'"

R is trying to save a file to a directory that does not exist. Why doesn't it exist? Well, in g.part1 GGIR checks if the top level meta data folder exists, and if it doesn't it creates it and all of the sub-directories. https://github.com/wadpac/GGIR/blob/c19ecd334c5d8e4119af5778d9bab40f56ca4106/R/g.part1.R#L49C1-L56C4

The problem is that you have created the first directory which passes the check, but not the subsequent directories. There are two ways to solve this:
* Manually create all the sub-directories that are required
* Use the shell function, which is the recommended way to interact with GGIR anyway and will take care of this for you. It's also what is shown in the vignette you link to.

So all you would be doing is:

```R
GGIR(
             mode=c(1), # This is the same as just running g.part1
             datadir="/Users/zhiyuan/Desktop/2024_8_26/Project/5/test",
             outputdir="/Users/zhiyuan/Desktop/2024_8_26/Project/5/out",
             ...
)
```

vincentvanhees, 2024-11-18:
Thanks @tarensanders 
@Chinate feel free to re-open if this does not address your issue.

closed 2024-11-18 by vincentvanhees