wadpac/GGIR #1225: Part 4: Error in if (is.na(defaultGuider)) { : argument is of length zero`

vincentvanhees, 2024-11-15:
Someone reported to me the part 4 error:
`1Error in if (is.na(defaultGuider)) { : argument is of length zero`
based on default GGIR settings.

I investigated it and see it is triggered by night number 0 to exist in the part 3 output, caused by revisions to function `g.det.sib` in the past year which made that night number 0 can be left in the data during the DST hour in the spring, which then is confusing for part 4 where night number is defined as the range in values in night number column of the part 3 output.

I am now working on a fix.

jhmigueles, 2024-11-18:
Hi, I have just observed the same and I was investigating this as well. Let me know if you need any help or you want me to double-check anything with another dataset. FYI, I observed this in the DST hour change in the autumn (not spring). 

It could be a good opportunity to revise the internal function `dstime_handling_check` within `g.sib.det`, I drafted some code for this in the branch "sleep_horAngle".

vincentvanhees, 2024-11-18:
I drafted a solution in https://github.com/wadpac/GGIR/tree/issue1225_DST_part3 and am waiting for Mike to confirm that this fixes the issue for him.

Explanation:

PR #1227 (2024-11-21): Fix handling of DST in the autumn in GGIR part 3.

Fixes #1225

Based on positive [feedback](https://github.com/wadpac/GGIR/issues/1225#issuecomment-2483527310) from Jairo I am now drafting this PR and probably merge it by the end of the week.

### Checklist before merging:

- [x] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [x] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [ ] Clean code has been attempted, e.g. intuitive object names and no code redundancy.
- [ ] Documentation updated:
  - [ ] Function documentati [... cut]
commit dd59a576 (2024-11-21): fixes #1225
closed 2024-11-21 by vincentvanhees