wadpac/GGIR #1229: In part 5, do.fragmentation produces duplicated ds_names

jhmigueles, 2024-11-21:
**Describe the bug**
`g.fragmentation` produces duplicated column names in output in part 5. This seems to occur when the window introduced to `g.fragmentation` has only one epoch. I observed this when calculating `qwindow = c(0, 6, 12, 18, 24)` in a recording that started at 18h. Then for segments 0:00-5:59, 6:00-11:59, and 12:00-18:00, `g.part5.definedays` incorrectly assigned segment indices of `segStart = 1` and `segEnd = 1`, and the rest of the analysis is done over these non-meaningful windows.

jhmigueles, 2024-11-21:
**UPDATE**

- Part 5 did not trigger any issue, but when extracting the reports for part 5, I got warnings indicating that some part 5 columns had been dropped from the report fos some files.
- This occurred in `g.fragmentation`, when the transition probabilities were skipped when not enough data points are available in a given segment (i.e., `Nepochs > 1`) => the proposed solution is to extract those metrics with value `NA` so that the number of extracted metrics and metrics names are always the same for all the files.
- Looking further, the reason why there were not enough data points available in those segments was because the segment start and end identification was not correct in `g.part5.definedays`. For example, if `qwindow = c(0, 6, 12, 18, 24)` and recording starts at 18:00, the segments 0-6, 6-12, and 12-18 are not available on the first day. Instead, `g.part5.definedays` was assigning a `segStart = 1` and `segEnd = 1` to these windows. Then, `g.fragmentation` was trying to extract fragmentation metrics over this single epoch, resulting in the issue described above.

vincentvanhees, 2024-11-22:
I am struggling to reproduce the issue. 

Can you clarify which columns you expect to be duplicated?
I am testing with a recording that starts at 17:15 and when I use:

```
qwindow = c(0, 11, 17.25, 24),
frag.metrics = "all",
```

I see in both "part5_daysummary_MM_L40M100V400_T5A5.csv" and "part5_daysummary_Segments_L40M100V400_T5A5.csv" 30 column names that start with "FRAG_" and none seem to be duplicated.

I also tried `qwindow = c(0, 11, 18, 24),` and `qwindow = c(0, 11, 17, 24),` but same result.

jhmigueles, 2024-11-22:
You are right, this is very specific to some of my recordings that have only 2 days recorded, and there are missing segments in the first and the second day. In this case, the transition probability fragmentation variables cause the issue. For example, recording starting at 18h on day 1 and finishing at 13:30 on day 2, qwindow = c(0, 6, 12, 18, 24).

**Day 1:** Segments 0-6, 6-12, and 12-18 are missing and transition probability outcomes are not calculated. Then, transition probability outcomes are calculated over segment 18-24, and ds_names are overwritten, compromising the naming of the outcomes for the first 3 segments (no errors/warnings observed).
**Day 2:** Transition probability outcomes are derived for windows 0-6, 6-12, and 12-18, but not for the last segment (18-24). As this is the last segment, ds_names are again rederived and the transition probability variables names are overwritten by the rest of fragmentation indices, producing the duplicate names in the output.

I think that the missingness of the last segment is what is actually causing the duplicate names. Here an example to replicate the issue:

```
# sample data
create_test_acc_csv(Nmin = 24*0.8*60, start_time = "18:00:00")

# run GGIR
GGIR(datadir = "123A_testaccfile.csv",
     outputdir = "./", studyname = "test",
     qwindow = c(0,6,12,18,24),
     frag.metrics = "all")

# load part 5 MM output
p5 = read.csv("output_test/results/QC/part5_daysummary_full_MM_L40M100V400_T5A5.csv")

# fragmentation metrics in output
frag_columns = grep("FRAG", colnames(p5), value = T)

# [1] "FRAG_Nfrag_PA_day"       "FRAG_Nfrag_IN_day"       "FRAG_mean_dur_IN_day"   
# [4] "FRAG_mean_dur_PA_day"    "FRAG_Gini_dur_IN_day"    "FRAG_Gini_dur_PA_day"   
# [7] "FRAG_CoV_dur_IN_day"     "FRAG_CoV_dur_PA_day"     "FRAG_alpha_dur_IN_day"  
# [10] "FRAG_alpha_dur_PA_day"   "FRAG_x0.5_dur_IN_day"    "FRAG_x0.5_dur_PA_day"   
# [13] "FRAG_W0.5_dur_IN_day"    "FRAG_W0.5_dur_PA_day"    "FRAG_SD_dur_IN_day"     
# [16] "FRAG_SD_dur_PA_day"      "FRAG_NFragPM_PA_day"     "FRAG_NFragPM_IN_day"    
# [19] "FRAG_CoV_dur_IN_day.1"   "FRAG_CoV_dur_PA_day.1"   "FRAG_alpha_dur_IN_day.1"
# [22] "FRAG_alpha_dur_PA_day.1" "FRAG_x0.5_dur_IN_day.1"  "FRAG_x0.5_dur_PA_day.1" 
# [25] "FRAG_W0.5_dur_IN_day.1"  "FRAG_W0.5_dur_PA_day.1"  "FRAG_SD_dur_IN_day.1"   
# [28] "FRAG_SD_dur_PA_day.1"    "FRAG_NFragPM_PA_day.1"   "FRAG_NFragPM_IN_day.1" 
```

PR #1232 (2024-11-22): Issue1229 duplicated frag names

Fixes #1229 => see issue. Changes applied to better handle partially available and non-available qwindow segments in the recording.

### Checklist before merging:

- [ ] Existing tests still work (check by running the test suite, e.g. from RStudio).
- [ ] Added tests (if you added functionality) or fixed existing test (if you fixed a bug).
- [x] Clean code has been attempted, e.g. intuitive object names and no code redundancy.
- [ ] Documentation updated:
  - [ ] Function documentation
  - [ ] Chapter vignettes for GitHub IO
  - [ ] Vignettes for CRAN
- [x] Cor [... cut]
closed 2024-11-22 by vincentvanhees