wadpac/GGIR #1235: Error in seq.default(rawTime[1], rawTime[length(rawTime)], step) : 'to' must be of length 1

gootofficial46, 2024-11-21:
**Describe the bug**
My GGIR script will not progress past Part 1 as the program seems to think that there is a problem with the timestamp data. It goes up by 1 second each row, is in the correct format and that format is specified inside my `params_rawdata` list. I also created a test .csv file with the same timestamp measurements but random accelerometer data to test the issue. I underwent multiple debugging options including testing timestamp parsing outside of GGIR as well as debugging seq.default outside of GGIR and every time the data showed an increase by 1 second and the functions worked with no isssue. There must be a problem with either my code or the GGIR shell function.

**To Reproduce**
Attached is the test .csv file and my GGIR script. It's my first time using GGIR so the code might look sloppy but all of the parameters should be correct. Run the test file through the code and you'll get the error outlined in the title: `Error in seq.default(rawTime[1], rawTime[length(rawTime)], step) : 'to' must be of length 1`

1. Sensor brand: GENEActiv
2. Data format: .csv
3. Approximate recording duration 7'days
4. Are you using a sleep diary to guide the sleep detection: No
5. Copy of R command used: `GGIR()`
6. Have you tried processing your data based on GGIR's default argument values? Does the issue you report still appear? YES

**R Code**
`dataFolder <- "/Users/georgeannas/Desktop/test"`
`outputFolder <- "/Users/georgeannas/Desktop/output"`

`# Load GGIR library
if (!requireNamespace("GGIR", quietly = TRUE)) {
  install.packages("GGIR")
}
library(GGIR)`

`# Check if output folder exists; if not, create it
if (!dir.exists(outputFolder)) {
  dir.create(outputFolder, recursive = TRUE)
}`

`# Run GGIR
GGIR(
  datadir = dataFolder,            # Input data folder
  outputdir = outputFolder,        # Output folder
  mode = 1:5,                      # Run all five GGIR modes
  overwrite = TRUE,                # Overwrite existing output
  do.imp = TRUE,                   # Imputation of non-wear time
  data_Format = 'csv',             # Specifies type of file
  chunksize = 1,                   # Number of days processed at a time
  params_rawdata = list(
    minimumFileSizeMB = 1,
    rmc.firstrow.acc = 2,
    rmc.firstrow.header = 1,
    rmc.header.length = 1,
    rmc.col.acc = c(2,3,4),
    rmc.col.temp = 5,
    rmc.col.time = 1,
    rmc.unit.time = "character",
    rmc.sf = 1,
    sep = ",",
    extEpochData_timeformat = "%Y-%m-%d %H:%M:%S",
    desiredtz = "VUT",
    deviceName = "GENEActiv"
    ),`
    
  `# Mode 1 (Pre-processing and calibration)
  do.cal = TRUE,                   # Perform auto-calibration
  do.enmo = TRUE,                  # Calculate ENMO (metric of acceleration)
  do.anglex = TRUE,                # Calculate angle relative to gravity
  windowsizes = c(1, 900, 3600),   # Indicates the short epoch, long epoch, and non-wear detection
  printsummary = TRUE,             # Print summary to console
  verbose = TRUE,`
  
 ` # Mode 2 (Non-wear time and sleep detection)
  do.part2 = TRUE,                 # Enable mode 2
  strategy = 2,                    # Sleep detection strategy (default is 2)
  hrs.del.start = 0,               # Ignore data before midnight
  hrs.del.end = 0,                 # Ignore data after midnight`
  
 ` # Mode 3 (Physical activity metrics)
  do.part3 = TRUE,                 # Enable mode 3
  threshold.lig = c(30, 100),      # Light activity thresholds (mg)
  threshold.mod = c(100, 400),     # Moderate activity thresholds (mg)
  threshold.vig = c(400, 1000),    # Vigorous activity thresholds (mg)`
  
 ` # Mode 4 (Individual reports)
  do.part4 = TRUE,                 # Enable mode 4
  do.visual = TRUE,                # Generate sleep and activity plots`
  
 ` # Mode 5 (Group summaries)
  do.part5 = TRUE                  # Enable mode 5
)`

`# Print message when done
cat("GGIR processing complete! Check the output folder for results.\n")`

**Expected behavior**
I expect GGIR to work through all 6 parts and produce me summaries and visualisation of the activity and sleep data recorded from the GENEActiv.

**Desktop (please complete the following information):**
 - OS: macOS 15.2 Beta (24C5079e)
 - GGIR Version: 3.1.5

[image]
[synthetic_data.csv](https://github.com/user-attachments/files/17853379/synthetic_data.csv)

gootofficial46, 2024-11-22:
Here is the code publicly available so you can download and test for yourself!
[](https://github.com/gootofficial46/GGIR_Practice)

vincentvanhees, 2024-11-29:
A 1 second time resolution not raw data based on GGIR's definition.
GGIR is designed to process data with multiple values per second, typical use cases involve 30 and 100 Hertz.
Such a high resolution is critical for separating movement from orientation component.

Has your data been exported from some other software with an option to increase the sampling rate?

vincentvanhees, 2024-11-29:
I see in the file paths George Annas, is this the same question as in https://groups.google.com/g/RpackageGGIR/c/2RTCUyby704/m/a9dCq4loCgAJ ??

Please try to avoid double posting. I am replying voluntarily. If you need my time urgently then just reach out with a paid consultancy request, I'd be happy to do that as I also need to pay my monthly bills.

closed 2024-11-29 by vincentvanhees